colloqi/piSignage #97: Path Traversal vulnerability

aegisduck, 2019-11-13:
# Describe the bug
### Suggested description of the vulnerability
: A path traversal vulnerability in the web application component of piSignage 2.6 allows a remote attacker authenticated as a low privilege user to download arbitrary files from the Raspberry Pi.

### Attack vector(s)
[image]
　1. Click the Log Download button at the bottom of the 'piSignage' administration page.  

[image]
　2. HTTP Packet is sent when the button is pressed.  

[image]
　3. Change the value of 'file' parameter to `../../../../../../../../../../etc/passwd`.  

[image]
　4. You can see that the /etc/passwd file is read.  

### Affected URL/API(s)
URL: `/api/settings/log`
Parameter: `file`

### Environment
* Raspberry Pi Hardware Version: Model 3B+ Revision: 1.3 Ram: 1 GB Sony UK
* piSignage Version: pisignage_2.6.1.img

colloqi, 2019-11-13:
Thanks, will sandbox the content, since it happens after login there is one level of check if you secure the webUI or disable altogether as of now

aegisduck, 2019-11-13:
Thank you for your answer. Are you saying it's not a vulnerability? Does this not apply to the next patch?

aegisduck, 2019-11-13:
> Thanks, will sandbox the content, since it happens after login there is one level of check if you secure the webUI or disable altogether as of now

Thank you for your answer. Are you saying it's not a vulnerability? Does this not apply to the next patch?

colloqi, 2019-11-13:
Yes and no, if the password is known, piShell is more dangerous place :) best way is to either block the webUI in such cases using server settings or change the password. However we will fix this vulnerability (it is) in the next release

colloqi, 2020-01-05:
Fixed in 2.6.4

closed 2020-01-05 by colloqi