colloqi/piSignage #98: Replace video

Nrozas, 2019-11-25:
There is an option that allows replacing a video with a new one in the assets area, so that only replacing it also changes it in the playlists.

If there was no possibility, it would be a good option to add to the system.

For example I have a specific offer that lasts 10 days in several playlists where I have it repeated several times among other videos, but within 10 days I will change it for another one, instead of uploading the new one and adding it to the playlists I would simply replace the previous video with the new one and it would be updated in the playlist without having one by one adding it again.

Thanks for any answer that helps me

colloqi, 2019-11-25:
Yes, it should work that way, but you need to press 
"Deploy All" or Deploy to groups.

Nrozas, 2019-11-25:
Thanks for your prompt response but I don't understand. What should work like this? Is there an option to replace one video with another? I can not find it?

colloqi, 2019-11-25:
Make sure the replacing video has the same name as original one, upload the video, it will overwrite

Nrozas, 2019-11-25:
>Make sure the replacing video has the same name as original one, upload the video, it will overwrite 

@colloqi Thank you so much, I never thought it was that simple.  A big greeting.
>Make sure the replacing video has the same name as original one, upload the video, it will overwrite 

@colloqi Thank you so much, I never thought it was that simple.  A big greeting.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

joelarcos, 2019-12-06:
@Nrozas where is this option to replace a video? I don't see it in the assets section.

hitesh83, 2019-12-07:
@joelarcos there is no option to click or a button to press.
Just upload new file with exact same name which is already in playlist.
Then Deploy, new file with same name will be sent to player.

closed 2020-01-05 by colloqi