colloqi/piSignage #100: iOS App

joelarcos, 2019-12-06:
I'm an iOS developer and would be interested in developing an open-source app for piSignage. 
I manage my players mostly on the road from an iPhone and I find that navigating the web portal on such a small screen is a little tricky at times. An app with a native iOS UI will provide a better user experience.
Is this something piSignage would be interested in?

hitesh83, 2019-12-07:
Native mobile app, or better simpler mobile UI .
I would be interested as a user.

joelarcos, 2019-12-29:
@hitesh83 Do you know whom I would contact to get info about API's?

hitesh83, 2019-12-30:
@joelarcos this is player app repo,
For managing Server side repo is - https://github.com/colloqi/pisignage-server
you can contact @ravibail for API details

colloqi, 2019-12-31:
https://piathome.com/apidocs/

joelarcos, 2020-01-03:
Great. I'll start the project and post a link here so that others can contribute.

joelarcos, 2020-01-29:
I have started working on the iOS app for piSignage. If anyone is interested in contributing here's the link: https://github.com/joelarcos/piSignage-iOS

closed 2020-07-30 by colloqi