colloqi/piSignage #104: Feature Request: Custom Fonts

Alladinsane, 2020-01-13:
It would be great if we could upload custom fonts as assets and then use them in messages, etc. I was able to accomplish this in a round-a-bout way on the managed hosting:
I create a custom template, and add the @font-face declaration as custom CSS that references the exact local path for the font on the player. Since the template is the "top-level" element. I can then reference the custom font in assets I create, and, once those assets get plugged into a playlist using the custom template, the font is found and used.

While this works, it's a lot of hoops to jump through. Also, since the open-source server doesn't seem to have the built-in template designer, I can't put my @font-face declaration there. Trying to put the @font-face declaration in the custom CSS of the asset where I'm trying to call it hasn't seemed to work.

Maybe I'm missing a much simpler work-around....

colloqi, 2020-01-14:
Thanks for the update, you could download the custom template from managed hosting, download the same and upload to open-source server to use it there.

Alladinsane, 2020-01-14:
Thanks! This is just exactly what I wound up doing. It sometimes means I need to go to the managed site and essentially recreate one of the existing premade templates in order to insert the custom css and download it, so it would still be nicer to have the ability to simply reference a custom font that exists as an asset. This is a workable solution, though. Really enjoying this signage system! Thank you!

Alladinsane, 2020-01-27:
I just wanted to revisit this, because I discovered a built-in feature I did not previously realize PiSignage had, and I thought it worth bringing up for anyone else who comes to this thread:

Basically, I came across this line in the documentation, "You can upload HTML file or zip repo with index.html as the entry point."

In other words, one can also include custom assets, like images and fonts, by dropping them into the same folder as the HTML doc, referencing them using relative paths, naming your HTML doc index.js, zipping it all up and uploading it to your server.

It's a testament to what a well-built system you've created that I keep finding new ways to throw content at it.

colloqi, 2020-01-28:
:)

closed 2020-01-28 by Alladinsane