colloqi/piSignage #109: Raspberry Pi 3 Model A+?

ersan, 2020-02-17:
Does anyone have this running on a Pi 3 Model A+?

I'm having a heck of a time getting it running.  It'll get to the welcome screen once but once I add the license/provision it, it reboots into the "loading" screen with the cookie and sits there for like 10 minutes - then the screen goes black and comes back completely white after another minute or so.

ersan, 2020-02-17:
I "fixed" my issues by commenting out "dtoverlay=vc4-fkms-v3d" in /boot/config.txt but I believe performance is somewhat reduced.

I'm not really a Pi expert but from what I can tell I think that line enables some sort of experimental OpenGL driver?

colloqi, 2020-02-18:
Can you try with 2.4.1 image? link is available in README file

ersan, 2020-02-18:
> Can you try with 2.4.1 image? link is available in README file

I mean yeah, that version works fine but probably because it doesn't have the vc4-fkms-v3d driver enabled at all.

hitesh83, 2020-02-18:
Hope the fact that a+ is having only 512mb of ram is nothing to do with this issue.

ersan, 2020-02-18:
> Hope the fact that a+ is having only 512mb of ram is nothing to do with this issue.

Eh it works fine on the Zero from what I understand, and when I disable that driver everything runs pretty well.

colloqi, 2020-07-30:
dtoverlay=vc4-fkms-v3d needs to disabled for certain cases like portrait mode

tw1xy, 2021-06-08:
> I "fixed" my issues by commenting out "dtoverlay=vc4-fkms-v3d" in /boot/config.txt but I believe performance is somewhat reduced.
> 
> I'm not really a Pi expert but from what I can tell I think that line enables some sort of experimental OpenGL driver? Seems to be some conflict with it and pisignage on the Pi 3 Model A+

Have you been running it since you opened this issue? Does it work well 24/7? Can you recommend buying Pi 3 Model A+ instead B+ or Pi 4?

Would be very helpful for me to know it since Pi 3 A+ costs half of B+. Thanks in advance :)