colloqi/piSignage #111: Pi 4 - video in 1080x1920 flickers (2.7.3)

akke911, 2020-02-26:
Video in 1080x1920px show lots of flicker. (tested multiple files)

Screen is native set up as portrait so groups -> settings -> orientation is landscape, but its the same flicker is on a testscreen set up as portrait. Resolution 1080p or auto does not make any difference on the flicker.

Video in 1920x1080 in the same group plays just fine.

[image]

bruxbl, 2020-03-03:
try 
sudo su -

apt update
apt full-upgrade

that fix problem for me 
but still p4 hang on 1080p after half hour of playing playlist

work well on 720p

Best,

akke911, 2020-03-04:
Thanks, but did not help in my case.

Problem seems to be related to resolution, resized video to 608x1080 works fine.

The p4 hang you have may be temp issues.

jimham1020, 2020-03-14:
sudo rpi-update fixed this issue for on rpi 4

akke911, 2020-07-02:
> sudo rpi-update fixed this issue for on rpi 4

Yes, this fixed the issue 🙂

closed 2020-07-02 by akke911