colloqi/piSignage #115: About piSignage Version an Raspberry Pi model

Nrozas, 2020-05-19:
I would like to have more sure which version of piSignage is valid for which model of Raspberry pi

For example I see that versions greater than 1.97d do not work on the raspberry pi model 2B, but I want to make sure which is the latest version that each model supports.

I can't find the list anywhere, we can use this thread to make this topic clearer.

Thank you very much for any help provided in this regard.

Nrozas, 2020-07-06:
Hi, Nothing about that??

colloqi, 2020-07-30:
Ideally this should be the case
1. 2.7.1 for Raspberry Pi 4 (and for Pi 3)
2. 2.4.1 for all other models

If 2.4.1 is not working on Pi 2B, let us know the error you are facing and we explore whether we can help. We actively test on Pi 3/4 and to some extent Pi zero