colloqi/piSignage #117: Glitch between HTML pages transitions

edpi, 2020-06-08:
Hi,

I have just downloaded 2.8.2 version on a RPi 3b, and created a playlist with two html pages. Group is configured with screen in 1080p resolution and portrait orientation, and html pages are loaded everytime they are shown (they contain real-time information, so it is not feasible to load them on memory).

The issue I have found is that transition between both pages takes too long, and during that transition it is shown a white glitch, which makes quite annoying to our customers.

Is there any way to avoid, or at least to minimize this effect?

Thanks

edpi, 2020-06-12:
Update on this issue:

I have tried on a fresh RPI 4 model, but the same problem is still there.

Regards

colloqi, 2020-07-30:
For real time cases, the page is loaded every time and currently it can not be avoided. But we are aware of this issue and looking out for solutions

Sh4d0wMaster, 2020-08-18:
I was running into a similar problem with the white flashing.  I switched from the Asset being a Web Page to being a Web Link and it loads and transitions much better.