colloqi/piSignage #122: unable to show still image in portrait mode with dual layout

roccopietrini, 2020-09-01:
Hi,
I'm trying to display an image and a video in portrait mode. I tried several image formats but none of them is being displayed, I only get the video displayed, with the other half of the screen that remains black. It works with an image in a single zone display portrait mode.
I've just updated the player to version 2.8.6 and launched rpi-config to update my raspberry pi 4. Nothing changes.
I think the problem could be related to the fact that I'm using the touchscreen 7" pi display, which is not connected via HDMI and thus doensn't send any EDID I guess.. My idea comes from the fact that if I choose the landscape mode with a video and an image, the video partially overlap the image, hence the image is not correctly positioned in his zone.. A workaround is to set in the group setting a resolution of 1280 (I noticed that this force the raspberry editing config.txt video option) which is not even correct for the display, that has a maximum resolution of 800x480. However this workaround doesn't work for the portrait mode issue I have.
If there is some additional information I should provide to get help, just let me know.
Thank you

hitesh83, 2020-09-02:
Hi, 
Just for clarification and for better understanding  putting below.
1- you got a RPi-4 with 2.8.6
2- your screen is not HDMI, so is it connected thru GPIO ? or RCA ?
3- your screen max resolution is  800x480
4- you want to run video / images in landscape mode .

now irrespective of resolution mention in PiSigange layout, you can set your video playing window in layout.
Even Pisigange layout is 1280x720, you can start video zone from 0x0 and with size you want.
Go to Playlist--->Layout-->Main Video Window-->here you can set Width and height of video zone and Left and Right offset.
By doing this your video should fit in the screen, for Image you can make a custom layout to suite your screen size.

Try this and let us know what output you get,

roccopietrini, 2020-09-02:
1- Exactly
2- DSI (It's the Raspberry Pi 7“ Touchscreen Display)
3- Exactly
4- No. I want to a portrait mode with a 2-zones layout (2bp layout), upper zone video and lower zone still image.

Problem is that video is showed in the upper zone, but image is black (using a single zone for the same image works, so it's not a matter of file type..). Changing the zones (putting image in the upper zone and video in the lower one) doesn't change. I will try to specify offset as you suggested and let you know.

roccopietrini, 2020-09-04:
I have to correct myself .. It does not work properly in portrait mode with a single zone layout image. I tried an image with resolution 720x1280 and even choosing stretched in the group settings, the image is not fullscreen but I have a black stripe on top of the image. A working solution is to switch the resolution from Auto (EDID) to 720x1280 (like my image).. but this resolution is not even the same aspect ratio of the display, which is 800*480 ! This can confirm that there are some issues when there is no EDID..

hitesh83, 2020-09-04:
So your issue is resolved ?

roccopietrini, 2020-09-05:
No it's not.. 2 zone portrait layout don't work, the image is not displayed..Have you ever tested it with the official raspberry 7" touchscreen display (DSI connector) ? I suspect it depends it is due to the EDID, which maybe is not sent by a DSI device, I don't know..

hitesh83, 2020-09-06:
1- even with EDID mismatch image should get displayed
2- personally I haven't tested it with DSI 
3- but with EDID mismatch I had tested working of image and video 
4- you might have to define image zone and video zone in custom layout
5- in single zone you are able to see image and video , so for two zone you need to define their respective positions.