colloqi/piSignage #126: Blank Screen on Book

Rebelturtle, 2020-09-30:
Pi boots and I see the green light flashing to indicate it is booting/reading from SD but the screen never changes from a blank screen.  Its connected via HDMI with pc monitor, I've attempted to Ctrl+Alt+T. I've re-downloaded the image again and re-written the SD multiple times.  I've pulled down a fresh image of other pi images and loaded them and booted without issues.  I'm at a loss on what to do next.  Uncle Google is not showing me anything new that I've not tried now
Using my test Pi which is a full Pi 3+kit with keyboard, mouse, power, SD, Pi, and case.  This should rule out the low power issues I've read about also.

anyone have any idea's or directions I can go to try and trouble shoot this?  I would really like to get this working as piSignage seems to be the better feature for what I'm looking for my work and my church.
I appreciate any information and assistance.
Thanks
RT

hitesh83, 2020-09-30:
Hi,
So to understand
1- You have RPi3 with SD card  
2- Screen is connected to HDMI
3- You burnt SD card with PiSignage image 
4- while booting you do not see anything beside blank screen. 

Now will need below info
1- is this RPi worked with Rpi default OS  or any other OS before ?
2- Sd card Class or speed ?
3- which version of Pisignage image you had used.
4-when you power on RPi and HDMI screen, do you see rainbow color screen ?
5- or what is the last screen you see before blank screen, do you see cookie eating animation ?

tunisiano187, 2021-09-14:
Same for me with Plasma screen since v2.9.9, OK on 2.9.8