colloqi/piSignage #127: Hardware rotated pi4 stuck on boot splash screen

andreacc81, 2020-10-02:
I have a pi4 rotated in group settings.
I can't make unattended updates or simply reboot because when it boot it get stuck on multicolor splash screen.
Never happened on non-rotated pi4s or pi3
Last time happened on version 2.9.0 (2.9.1) still not available on my on prem server

colloqi, 2020-10-03:
It needs a firmware update, can you check "vcgencmd version" and firmware version is Sep or later. You could update using

sudo update;sudo apt full-upgrade -y & in ssh or piShell. In case it does not get updated you could try sudo rpi-update as last measure

andreacc81, 2020-10-05:
Thanks.
Firmware was Jan 30 2020 13:50:45 
Dia apt-get update & apt-get full-upgrade and it updated to Sep  2 2020 21:13:13 
It also updated more than 140 packages. Is this normal? Consider that i was on version 2.9.0.
If i undestrand correctly when deploying a new player i need to manually update packages, being on latest pisignage version is not enough?
I'll try to reboot the player sometimes and i'll let you know if problem is solved.
Thanks

andreacc81, 2020-11-11:
Just had same issue also on non rotated pi4. Confirm that it does not happen on latest firmware

closed 2020-11-11 by andreacc81