colloqi/piSignage #128: Player created with 2.9.0 image fail to update to 2.9.1

andreacc81, 2020-10-06:
New player created with 2.9.0 image does not update to 2.9.1.
When update is triggered from the (on prem) server the player say that is updating, reboot, but the version remain 2.9.0

andreacc81, 2020-10-06:
[forever.log](https://github.com/colloqi/piSignage/files/5334545/forever.log)

Log attached

hitesh83, 2020-10-06:
It is working for hosted players on pisignage.com 
opensource need to be checked.

andreacc81, 2020-10-06:
I use opensource on premise server. It is not working as you can see from log

hitesh83, 2020-10-15:
Update your server now with "git pull origin master"
and try to update player from 2.9.0 to 2.9.2
It is working now.

[image]

andreacc81, 2020-11-11:
Thanks. Is working with same version of server.
Unfortunately updating server caused loss of communication between server and all players with version prior to 2.9.2, had to update them manually through player web interface