colloqi/piSignage #131: Playlist not starting over

zifnabbe, 2020-11-20:
I've recently created a new playlist (a week ago), but it doesn't repeat when it arrives at the end. It just contains two assets.
One radio stream (in background) and one assets which links to a published google slide show.
After 620sec it should start over, but it doesn't. I've set this up for two players and both have this issue.

versions of pisignage:
ver: 2.9.3/
ver: 2.9.3/stretch_9.6_admin_2019-01-30

One is a recent Pi 4, B model. the other is a Pi 3, Model B

An 'old' playlist (of few months ago) doesn't have this problem. It repeats when it arrives at the end.

I'm not sure were to look at or what might be the issue here.

Any hints? Am I missing something obvious here?

Thanks!

hitesh83, 2020-11-20:
Hi,
Your "old" playlist still work as expected on current players with version 2.9.3
And was there any version difference between working and non working loop?
Thanks

zifnabbe, 2020-11-20:
yes, the older playlist works on v2.9.3. Settings are identical.

I was just realizing the following: 

The second asset is just an html page which links to a google slide that starts the show. With a repeat, the asset is just the same as the one already displaying (if we don't take the radio asset into account). Might it be that this asset is -in this case- not re-initialized (and that way the google slide is not restarted?)

The older playlist -which works fine- also uses this technique, but in this playlist I user several of these assets and vids, so in that case the 'next' asset is not really the same as the 'previous'...

I've just tested this with the 'reload link URLs each time' option 'on', and now it works as it should. I was not sure if html assets would be effected by this option, but it seems that they are.

In the older play list, the 'reload link URLs each time' option is 'off'. So there is some behavior difference... or nor?

Anyway, for me the is case solved. 
If there is some behavior difference, than it might be interesting to look at it further.

Sorry for bothering you all.

(another option would be to set the loop option in the google slides to true)

closed 2020-11-20 by zifnabbe