colloqi/piSignage #133: read-only raspberry

roccopietrini, 2021-01-12:
Hi,
Has anyone experience on making the raspberry read-only ? I would like to do this to minimize SD card corruption issues..

hitesh83, 2021-01-13:
Pisignage images are very lite on sd card,  we are running it almost 18hr daily for past 3 years in 100+ installation.
Till date haven't face any SD card issue.

As we keep changing our content every alternate day, making SD read only is not a good idea for us. 

For small local install with no change in content, read only should work.

roccopietrini, 2021-01-13:
Do you schedule poweroff / poweron on the raspberry everyday? It is my understanding that incorrect poweroff can damage the SD card (can I also ask if you recommend a card vendor?)..Thx

hitesh83, 2021-01-13:
We do not schedule ON / OFF in Raspberry pi,
ON/OFF is controlled through a timer installed in the power distribution box.
SD card may get damaged due to incorrect power off, but due to very low read-write operations, things are still good for us.

Mostly we used SanDisk Class 10.

[image]

roccopietrini, 2021-01-13:
Thanks! Just to confirm.. you don't shutdown the raspberry (with shutdown or poweroff command in raspbian) before the power loss.. is that correct ?

hitesh83, 2021-01-13:
yes, your understanding is correct, 
actually, it is not the correct way for RPi, but digital signages need to run like this, and we are ready to replace the SD cards when it fails.
And there are many if and but in running a script on time to shut down, so we are taking the risk.

closed 2021-01-14 by roccopietrini