colloqi/piSignage #136: Weblinks not cached

ezraholm50, 2021-02-20:
Hello! I've just bought 9 licenses to use with my self hosted pisignage server.

Everything works except that weblink pages do not get cached, when i login and the page refreshes it logs me out again.

Also there is no option to disable the refresh of an asset when there is only 1 asset in the playlist. I cannot set the rotate seconds option to more then about 3 weeks (converted to seconds ofcourse), i just like to disable it for those.

I have 4 of the 5 PI's showing a static webpage where we are logged in to a service and they must remain that way as long as possible. 
Those websites do not offer api/url additions to specify for authentication.

Also i was wondering if anyone researched the option for a VNC connection to the chromium-browser-v7 that the players use?
Since there's no desktop installed this is hard to do. This would allow remote login once in a while.

please advise.

ezraholm50, 2021-02-20:
Ok i managed to get the VNC part working, for anyone that also would like to have this:

Create a script:
`nano vnc.sh`

Paste content and change VNCPASSWORD=changeme:
```
#!/bin/bash
VNCPASSWORD="changeme"

if [ "$EUID" -ne 0 ]
  then echo "Please run as root: sudo bash vnc.sh"
  exit
fi

cat << VNC > "/etc/systemd/system/x11vnc.service"
[Unit]
Description="x11vnc"
Requires=display-manager.service
After=display-manager.service

[Service]
ExecStart=/usr/bin/x11vnc -xkb -noxrecord -noxfixes -noxdamage -display :0 -auth guess -rfbauth /etc/x11vnc.pass -o /var/log/x11vnc.log -rfbport 5900
ExecStop=/usr/bin/killall x11vnc
Restart=on-failure
Restart-sec=2

[Install]
WantedBy=multi-user.target
VNC

systemctl daemon-reload
systemctl enable x11vnc

apt update 
apt install xfce4 x11vnc xinit xserver-xorg -y
x11vnc -storepasswd "$VNCPASSWORD" /etc/x11vnc.pass
export DISPLAY=:0

reboot
```
Execute script:
`sudo bash vnc.sh`

I use this on a RPI4 with the latest Pisignage Player with 1 monitor connected to the first (main) HDMI out on the PI.

colloqi, 2022-04-20:
Hi

Server settings override player settings, can you change it under Group settings of server UI and check?