colloqi/piSignage #137: [new function]  Master- Client in player

hitesh83, 2021-03-09:
In a group of players inside the Local Area Network
One player to act as a master, rest to act as a client
Master will broadcast instructions on UDP which advert/assets to be played by the Client player. 

This will be useful for synchronized play, 
use case 
1- Cinema hall lobby 
2- Art Gallary installations
3- Passenger information display in metro / Railway / Airport etc...
4- Corporate or small to large functions
5- Malls / Hospitals etc..

nomadixmedia, 2021-03-16:
You can do this already with the server. Just deploy on a device on the local lan. You can also run server on a PI.

hitesh83, 2021-03-17:
I am not aware of such functionality in either hosted or opensource server.

I am talking about frame to frame sync between group of players.