colloqi/piSignage #142: Livestreams from Youtube or simple Mjpeg-Streams are no longer displayed

Stephan-PS, 2021-06-07:
Hi Folks,

thanks for reading my entry - i hope you can help me to fix this problem.

Since i updated the Pisignage Player to 2.9.6 and newer, the player dont show anymore Youtube Streams from Webcams or Live-Streams from TV Stations.

How can i fix the problem or must i setup the player new with an older Player-Software lower than Version 2.9.6?
Downgrade ist not available?

Thanks for your help.

Best Regards
Stephan

glen4cindy, 2021-06-12:
I began having the same problem. I ended up starting out with a new SD card and downloaded the 2.4.1 from here and then let the system automatically update itself.

I'm currently at 2.9.9 and YouTube streams are still working.

Stephan-PS, 2021-06-14:
Hello Glen, 

when i remember right i startet with 2.7.1 and updated the Player Software, when a new Version was shown. 
Since Version 2.9.6 no live-Link (mjpeg from an webcam source) or YT-Stream (Webcams TV-Stations) is played anymore and the links are still working on other platforms.
Any ideas or reasons why this problem occurs?

Best regards
Stephan

hitesh83, 2021-06-15:
might be when you update from 2.7.1 to 2.9.x , only player software is updated and not the operating system and other supporting software, you can do that manually via PiShell.
thanks
Hitesh

Stephan-PS, 2021-06-21:
Hello Hitesh, i will try in the next days. 
Thank you.
Stephan

Stephan-PS, 2023-07-27:
Currently since version 3.2.7 again, no Youtube Livestream would be played e. g. EuroNews Livestream (the Stream works in every Browser and before upgrading to 3.2.7 everything was fine runs over a lot of month without problems. 
Did someone test version 3.2.8 ?
I choose Livestream .TV an add this link "https://www.youtube.com/watch?v=CQ3KsEbsYHs?&autoplay=1" runs in the pas without any problem.
Did someone have an hint to fix this?
Best Regards

geevihegde, 2023-07-28:
This issue can be solved by updating yt-dlp program as follows.
`sudo yt-dlp -U  &`

For more details, Please see https://github.com/ytdl-org/youtube-dl/issues/31530