colloqi/piSignage #143: Weather Example CORS Error

Scifiknux, 2021-06-13:
Getting this error when loading weather example:

`Access to fetch at 'https://pisignage.com/api/getweather?place=*****&units=imperial&lang=en' from origin 'null' has been blocked by CORS policy: Response to preflight request doesn't pass access control check: No 'Access-Control-Allow-Origin' header is present on the requested resource. If an opaque response serves your needs, set the request's mode to 'no-cors' to fetch the resource with CORS disabled.`

https://github.com/colloqi/piSignage/blob/master/examples/weather/weather.html

msmedien, 2021-06-14:
Same problem here. I designed a own weather-widget, based on the code of the pisignage weather example. It dosent work anymore... A quick solution would be awesome. :)

hitesh83, 2021-06-15:
If I replace those ***** to some city, I get JSON response in a desktop browser, 
So the widget should work, did you guys tried opening this URL in a desktop browser?
e.g. : - https://pisignage.com/api/getweather?place=Pune&units=imperial&lang=en

right now don't have RPi connected so not confirming the same on actual pisignage.

msmedien, 2021-06-15:
Hi @hitesh83, thanks for your answer. :) The widget is works again... Maybe a short server problem from openweather? If you like you can check my widget under: http://widgets.altmarkdisplays.de/altmarkdisplays/weather/

The widget is not finished yet. It will work best under a 1920 x 1080 px resolution in fullscreen. In the future it will be responsiv for different resolutions. ;)

Thanks for the help! :)

hitesh83, 2021-06-15:
yes widget is working fine at my end too, looking good !

Scifiknux, 2021-06-15:
> If I replace those ***** to some city, I get JSON response in a desktop browser,
> So the widget should work, did you guys tried opening this URL in a desktop browser?
> e.g. : - [...]

Hi @hitesh83 I should have specified but I replaced my zip code with the asterisks for confidentiality. Even with the zip/city filled in correctly, it wasn't working. This seems to have been resolved now though. Thanks for your time, all!

closed 2021-06-15 by Scifiknux