colloqi/piSignage #144: Image flickering when player is turned on for longer than one week

mathidepauw, 2021-06-14:
I experience serious flickering when the player has been playing for more than one week. The first week there's almost never an issue but after that the chance of flickering increases substantially.

With flickering I mean a one second black frame is shown every 3-5 seconds. This only disappears after a reboot of the pi.

I set up a putty script to reboot every couple of days, which is a good workaround but just wanted to let you guys and gals know!

Scifiknux, 2021-06-14:
@mathidepauw I have almost 30 Raspberry Pis running the latest PiSignage and none of them are showing this issue. Can you give any more information? What type of Pi are you running? How are you flashing the software to the SD card? What video connections are you using?

mathidepauw, 2021-06-15:
@Scifiknux I'm using a rpi 3B + with a Sandisk Ultra 32 GB microSD. I used Balena Etcher to write the ISO and use HDMI (not sure about the version though).

But if you have 30 and never experience this, I guess it's not PiSignage then. Back to the drawing board for me.

Scifiknux, 2021-06-15:
That's so odd. I have a couple ideas for troubleshooting though:

What content is in your playlists? Does the flickering occur on all assets, or just one type (like video)?

Also have you tested basic Raspbian/Raspberry Pi OS on those screens? Wondering if there may be an incompatibility with the RPi's framerate and the TV's. Might be worth checking the TV's settings as well.

Lastly how's your power supply? I have both Pi 4s, 3s and 3b+s in my fleet but the 3s sometime flicker if they're not getting enough power.

Hope this helps in some capacity!