colloqi/piSignage #146: Players unresponsive, ping ok, ssh and web if ko

andreacc81, 2021-07-13:
I have 2 players (in a remote branch office...) registered on onprem server that at least once a week are unresponsive, red dot in players menu on server, ping is ok, ssh and web interface are not respondig.
They come back online with a power cycle.
They are updated to version 2.9.9 running on pi 4 4gb.
When the come back online i can't find anything useful in logs

andreacc81, 2021-07-14:
[image]
I set daily reboot at 00:00, one of the two machines went unresponsive 2h and 30min after reboot.
Still pinging but unmanageable.
Really need to solve this issue

colloqi, 2021-07-14:
Try upgrading the Raspbian OS

```
sudo apt update
sudo apt full-upgrade -y 
```

andreacc81, 2021-07-21:
Both upgraded as instructed
Still becoming unresponsive after a couple of days

colloqi, 2021-07-26:
Please send us the player log /home/pi/forever_out.log at support at pisignage.com

GT3CH1, 2022-02-07:
I found that updating from version 2.9.x to 3.x.x broke the client completely, as my server didn't fully download an update, thus causing the unzipping of the new update to fail.  I had to copy the piSignagePro folder from one client to another to resurrect it.