colloqi/piSignage #148: No connect to on Primise after Upgrade to 3.0.1

solunix, 2021-07-23:
Tested with fresh new 2.9.8 Image, working fine, then Upgrade to 3.0.1 on a Pi Zero

Forever.out throws
error: socket.io: error  event,,,, reserved fields must be empty
five times in a second

speedboatz, 2021-07-25:
Same here. I updated my selfhosted server by issuing git pull origin master, but to no avail. Every player using v3.0/v3.01 becomes unresponsive

colloqi, 2021-07-26:
There are two socket.io versions being tried (new - 2.x and old 0.9.19). 2.x has been moved completely to websocket in 3.0.1 and if that is blocked/not working this issue could happen. Can you post parts of forever_out.log where new socket.io error is reported?

solunix, 2021-07-26:
pisginage-server branch is up to date with 'origin/master' and latest changes from last Friday
npm install up to date, audited 574 packages

`Express server listening on port 8000
7/26/2021, 6:48:27 AM - info: Spawning wget -rNnHd,--follow-tags=a,--no-parent,--no-check-certificate,-t,20,--waitretry,60,--retry-connrefused,--no-cache,-R,PlayerName,http://mylocalserver:3000/sync_folders/myaccount/PlayerName; Running as PID 910
7/26/2021, 6:48:27 AM - info: cpu serial number: 900000xxxxxxxxxxxxxx
7/26/2021, 6:48:27 AM - info: license status: true
7/26/2021, 6:48:27 AM - info: MAC Address : xx:xx:xx:xx:xx:xx
7/26/2021, 6:48:28 AM - info: Screen resolution from tvservice -s: 1920,1080
7/26/2021, 6:48:28 AM - info: resolution changed to 1080p,orientation set to : portrait270,edid resolution is : 1920,1080
7/26/2021, 6:48:28 AM - info: youtube-dl is available at /usr/local/bin/youtube-dl
7/26/2021, 6:48:28 AM - error: Error in setting/deleting cron job; Error: Command failed: sudo crontab -l
no crontab for root
;no crontab for root
sudo crontab -r
7/26/2021, 6:48:28 AM - error: Error in setting/deleting cron job; Error: Command failed: sudo crontab -r
no crontab for root
;no crontab for root
7/26/2021, 6:48:28 AM - info: wget stopped with code 0 and signal null; downloadInProgressCount: 1
7/26/2021, 6:48:28 AM - info: COPIED SERVER FILES PlayerName SUCCESSFULLY.
7/26/2021, 6:48:29 AM - info: starting socketio to http://mylocalserver:3000with ver: new
7/26/2021, 6:48:29 AM - error: socket.io: error  eventundefined;Error: websocket error
7/26/2021, 6:48:29 AM - error: socket.io: disconnect  eventundefined;transport error
7/26/2021, 6:48:29 AM - info: socket.io works with 0.9.19
7/26/2021, 6:48:29 AM - info: Chromium is available at /usr/bin/chromium-browser
7/26/2021, 6:48:29 AM - info: Starting websocket
7/26/2021, 6:48:29 AM - info: Openvg is available at /usr/local/bin/openvg_display
7/26/2021, 6:48:30 AM - info: Openvg is *not* available as it is buster(pi 4) OS release
7/26/2021, 6:48:30 AM - info: Number of display units: 1
7/26/2021, 6:48:30 AM - info: evince pdf reader is available at /usr/bin/evince
7/26/2021, 6:48:30 AM - error: socket.io: error  eventWUlqjsG7Ii98YAIUAFOs;Error: websocket error
7/26/2021, 6:48:30 AM - error: socket.io: disconnect  eventundefined;transport error
7/26/2021, 6:48:30 AM - info: Browser loading /home/pi/piSignagePro/templates/1.html; Running as PID 1015
7/26/2021, 6:48:30 AM - error: socket.io: error  eventj4Ot-YgKomdv5hdnAFOu;Error: websocket error
7/26/2021, 6:48:30 AM - error: socket.io: disconnect  eventundefined;transport error
7/26/2021, 6:48:31 AM - info: browser stderr: BlueALSA detected - Disabling audio sandbox
7/26/2021, 6:48:31 AM - error: socket.io: error  eventl8vQNxyddh9I33_sAFOw;Error: websocket error
7/26/2021, 6:48:31 AM - error: socket.io: disconnect  eventundefined;transport error
[...repeating 3 times with different random chars...]
7/26/2021, 6:48:35 AM - info: ces support: opening a connection to the CEC adapter...
requesting CEC bus information ...
CEC bus information
===================
device #0: TV
address:       0.0.0.0
active source: no
vendor:        Samsung
osd string:    TV
CEC version:   1.3a
power status:  on
language:      ger
device #1: Recorder 1
address:       1.0.0.0
active source: no
vendor:        Pulse Eight
osd string:    CECTester
CEC version:   1.4
power status:  on
language:      eng
currently active source: unknown (-1)
;;null
7/26/2021, 6:48:35 AM - info: cec IS supported
7/26/2021, 6:48:36 AM - info: browser stderr: libEGL warning: DRI2: failed to authenticate
7/26/2021, 6:48:36 AM - error: socket.io: error  eventEySnECpSJ5Ku-4FuAFO4;Error: websocket error
7/26/2021, 6:48:36 AM - error: socket.io: disconnect  eventundefined;transport error
7/26/2021, 6:48:38 AM - info: browser stderr: [1069:1069:0726/064838.702162:ERROR:viz_main_impl.cc(229)] Exiting GPU process due to errors during initialization
7/26/2021, 6:48:39 AM - info: browser stderr: [1015:1138:0726/064839.238694:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
7/26/2021, 6:48:39 AM - info: browser stderr: [1015:1138:0726/064839.248569:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
7/26/2021, 6:48:39 AM - info: browser stderr: [1015:1138:0726/064839.253338:ERROR:object_proxy.cc(621)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
7/26/2021, 6:48:39 AM - error: socket.io: error  event_JT-YSjJ4bAAyewwAFO8;Error: websocket error
7/26/2021, 6:48:39 AM - error: socket.io: disconnect  eventundefined;transport error
7/26/2021, 6:48:40 AM - info: browser stderr: [1113:7:0726/064840.288367:ERROR:command_buffer_proxy_impl.cc(122)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
7/26/2021, 6:48:40 AM - error: socket.io: error  eventcgP4o0q-wIMPkIs_AFO-;Error: websocket error
7/26/2021, 6:48:40 AM - error: socket.io: disconnect  eventundefined;transport error
[... cut ...]
[...hundred times with different random chars...]
7/26/2021, 7:08:44 AM - warn: Restarting wlan interface
7/26/2021, 7:08:53 AM - error: **** request error, please check RSS feed URL http://myfeed/rssfeed.xml
7/26/2021, 7:09:14 AM - info: starting socketio to undefinedwith ver: 0.9.19
7/26/2021, 7:09:14 AM - info: socket.io: connected to server yyEQIdeWoBDB27HG55v2with ver: 0.9.19
7/26/2021, 7:09:14 AM - error: socket.io: error  eventyyEQIdeWoBDB27HG55v2;reserved fields must be empty
[...hundred times with different random chars...]`