colloqi/piSignage #155: missing change logs

si458, 2021-11-08:
ive updated twice in the past 2 weeks from 3.0.4 to 3.0.5 to 3.0.6

the is no change logs for what these changes are

it would be nice to update the changelog more frequently
OR
show the changes on the web panel when you click the upgrade icon

wrldtvlr, 2021-11-10:
Agreed - would really be helpful to understand changes.

si458, 2021-11-10:
@wrldtvlr the release notes where just updated yesterday, 
however i do agree they are slow and it might be better to list the changelog online when you do the upgrade

colloqi, 2021-11-11:
Sorry, we skip/delay updation of release notes when the changes are minor but urgent fixes. Will try to avoid this as much as possible in the future!

closed 2024-07-27 by si458