colloqi/piSignage #156: Syncing 3 Screens Together

CapturedSecurity, 2021-11-12:
I have three Raspberry Pi 4's that start displaying the customer's menu at a certain time. While displaying the menu there is an advert that plays in sync but for some reason the more the ad plays the more they get out of sync the three TVs get. I checked all three videos and they are all 6 seconds long. What else could be causing this issue?

davidecat, 2021-11-12:
Hi, 2 or more units will never be on sync only cause the contents plays at the same time and they have the same lenght. Many other variables impact, like other processes running, different access time to the microsd card, etc...  Everytime a new instance of omxplayer is launched it takes different time in each of the units... In my experience I tell u it will never work for long time... U can try to minimize the issue setting a change of playlist more often, like every 10 minutes... this will reduce the difference.. but If u are looking for final solution u must have a master that broadcast one single stream and a client that plays...

hitesh83, 2021-11-12:
I have been using pisignage for many years, had faced this issue at various locations, playlist start time is the same for all players, they also start at exact same time, but due to reasons outside pisisnge content loses sync after a few minutes. 
this is mainly IO getting affected by RPi OS.
to overcome this issue, I had to write a custom script for broadcast to the client.

There are two ways to get the perfect sync., hope one of the methods is adopted by pisignage soon.

1- Video-only sync - This gives me the perfect frame to frame sync using UDP multicast in the local network. but limited to video content ad no html page/url.

2- All content sync -, this is not a frame to frame sync, but very close to the frame to frame sync, this uses master sending instructions to players what to be played for every asset in the playlist.

I am using both methods as per requirement, both are possible with PiSiange with external help.

davidecat, 2021-11-12:
> I have been using pisignage for many years, had faced this issue at various locations, playlist start time is the same for all players, they also start at exact same time, but due to reasons outside [...]
Could we know amore about this "external help"? :-) 
Thank you!

hitesh83, 2021-11-12:
@davidecat this needs to be developed at your end as this is not a part of PiSigange or PiSiange team need to take this in to do list, 
I had done proof of concept, also made a makeshift utility of my own use in python and shell scripts.

jebbett, 2021-12-23:
Hi, I'm also having this problem as I have matched content between screens, a regular API call helps but isn't really the solution.

For the playlists you already know the expected duration of each slide, so what you could do is take the start time put that in to a variable and then just continue to add the slide time to it, and schedule the next slide change for that time rather than say 60 seconds from the last change.. if you need to set a number of seconds rather than an actual time then you just need to do the math subtracting one from the other.. Hope that helps, I haven't looked at your source to see exactly how you are doing it currently.