colloqi/piSignage #157: License issue... Player not showing under Registered player....

saintofinternet, 2021-11-13:
License issue... Player not showing under Registered player.... but

the Licensing page is showing as 1 License is active.... I am unable to recover that license.

[image]

[image]

[image]

hitesh83, 2021-11-13:
License generated as "Player License Only"  are not shown in player section.
These License are for open sources server users .

If you want to manage your player with pisignage.com, then de-register exiting player by going into settings and then again register using "managed" open while registration of player.
[image]

saintofinternet, 2021-11-14:
so how do i remove the specific license and re-register it if I cannot manage it under the player section?

hitesh83, 2021-11-14:
Deregister option is there under settings, 
Go to settings in pisignage.com, scroll till bottom , 
And register it gain by selecting managed option.

saintofinternet, 2021-11-14:
precisely that license is not visible under that section...

[image]

saintofinternet, 2021-11-14:
and after all this the bottom of the screen is still showing the " This player is powered by pisignage.com " message.

Samuel12321, 2023-05-21:
This is still broken, any eta on a fix

AbhijithBn, 2023-05-21:
Hi @Samuel12321 , please write to us at support at pisignage.com mentioning your account name and we can help you with it.

Samuel12321, 2023-05-21:
Thanks for the prompt response @AbhijithBn , your team has already unlinked my devices once after i emailed for support. However as i am evaluating pi signage for use in a business i need to be able to quickly and easily link and unlink testing devices to my account, so emailing for support every time isn't practical. I also need to be able to demonstrate that pisignage is reliable and fulfils all our operational requirements.
Is there any ETA on a fix for the "de-register a player" menu?

AbhijithBn, 2023-05-21:
@Samuel12321 
Deregistering a Managed Player works from Server UI from the Settings page.
However, for Players on the Open Source server(registered through Player Only License), we would want to keep control over deregistering the Player to avoid misuse of Licenses.