colloqi/piSignage #159: Black Screen after several hours on two different devices

jebbett, 2021-12-02:
Hi, I have two Pi's both running the latest software, originally installed from pisignage_3.0.4.img from your site, playing photos and a few videos, both screens have went blank but have been running for several hours today fine, but have just turned around to see both of them with black screens, the screens do not appear to be off, just black as there is still backlight. I'm not sure if both were at the exact same time but both were running about 6 hours fine.. 

1. Raspberry Pi 3 Model B Rev 1.2
2. Raspberry Pi Zero 2 Rev 1.0

Both connected to LG monitors, also tried plugging in to a Dell screen but no change. 

I've also tried:
1. export DISPLAY=:0 then xset s reset with no effect
2. Plugged in a USB keyboard/mouse still nothing
3. Tried to change playlist to TV off and then back.. 
4. Only a reboot via command line appears to fix it

Any help appreciated

hitesh83, 2021-12-02:
If you can upload forever logs from player, that would be helpful to know the issue.

jebbett, 2021-12-03:
Logs attached, but I don't think they show any issue.

I expect the screen became blank some time around 13:00 (after it was turned back on at 12:18), I'll try removing the videos from the playlist and see if that points to the issue

[forever_err.log](https://github.com/colloqi/piSignage/files/7650160/forever_err.log)
[forever_out.log](https://github.com/colloqi/piSignage/files/7650161/forever_out.log)
[forever.log](https://github.com/colloqi/piSignage/files/7650162/forever.log)

jebbett, 2021-12-07:
To update on this issue.

The video possibly causing the issue was uploaded as a H264 MPEG-4 AVC, but I had changed the video player from default to MPV to avoid a black screen between videos.

It seems that MPV struggles, which is a problem as I have some back to back videos and without this I get a black screen between the videos, even just playing the video normally I'm getting some small stutters, so any advice on the codec I should use for perfect playback?

Playing on default seems to work, on MPV on the most recent error I found it at the login screen (but I couldn't type anything despite a connected keyboard) and the web portal was still showing it as playing but said the HDMI was off (which it wasn't).

Hopefully that gives you something to work with, happy to share the specific video if that is helpful.

I've reverted to default in the mean time and appears to be running fine.. I can only assume either the CPU is being taxed in the merge of the videos or it's a memory leak perhaps..