colloqi/piSignage #165: Custom Grid Layout - Video playlist playing on wrong part of screen

rt050, 2022-01-18:
Hi there,

I've made a custom layout utilizing the handy tool on PiSignage website.

After some careful tweaking I was able to get it to perform how I wanted for my display. Essentially the screen is split into 1 x "half width x full height panel" on the right hand side with "2 x half width x half height panels" on the left of the display.

Whilst this works (playlists etc) and displays correctly with image and web page assets, as soon as I bring an video into the mix, the video plays on the opposite side of the screen to where it should be.

If I move the video asset to the other zone (e.g. top zone instead of the bottom zone), it still plays on the other screen but moves to the vertical positions of the zone.

Would anybody have any pointers? I'm a bit confused as to what could be causing this.

Custom layout code is attached. Sorry, couldn't attach it in html so it's in a txt file.

Thanks :)

[custom layout.txt](https://github.com/colloqi/piSignage/files/7890713/custom.layout.txt)

colloqi, 2022-01-20:
You could try the following and see
1. Add the same co-ordinates under main and side/bottom video window sizes under layout popup
2. Select 1920x1080 instead of auto (and optionally edit /boot/config.txt for your specific resolution if you are brave!)
and  Select portrait mode

rt050, 2022-01-20:
> You could try the following and see
> 
> 1. Add the same co-ordinates under main and side/bottom video window sizes under layout popup
> 2. Select 1920x1080 instead of auto (and optionally edit /boot/config.txt for your specific resolution if you are brave!)
>    and  Select portrait mode

Hi Colloqi,

Thanks for your reply. Can I just ask what these are in relation to?

Bullet point 1 - is that under the default layouts as opposed to the custom layout that I attached the config for?

Bullet point 2 - do you mean change the res in the player and/or group settings?

Thanks

colloqi, 2022-01-21:
Bullet point 1 - is that under the default layouts as opposed to the custom layout that I attached the config for?
>> it is an attached screen (next 2 tabs) under layout popups

Bullet point 2 - do you mean change the res in the player and/or group settings?
>> Group settings