colloqi/piSignage #166: No Audio on Aux when selected mpv under group settings

XenHuang, 2022-01-24:
Hi
I am trying to stream YouTube video and I want the sound to output on the 3.5mm jack.
Is there way I can get audio from my 3.5mm jack when I select mpv player under group setting?
It only has audio output on hdmi, but I need it on the 3.5mm jack only.
I know omxplayer has audio on 3.5mm jack, but the sound quality is poor due to it’s defaulted to output on both hdmi and jack. I am guessing it’s because pisignage is using 
omxplayer -o both “path” 
to execute the play command, is there a way I can change it to 
omxplayer -o local “path”?

hitesh83, 2022-01-25:
Have you tried setting audio route ?
sudo amixer cset numid=3 1
Last number defines audio hardware to use
1= 3.5mm jack
2=hdmi
0= auto

colloqi, 2022-01-25:
You could also use sudo raspi-config and select the audio output

XenHuang, 2022-01-25:
Thanks for the comment. I tried both of these, but I am still not getting audio from 3.5mm jack when using MPV ( checked "use MPV to avoid gap between videos" under group setting). If I unchecked "Use MPV to avoid gap between videos" to use omxplayer, I get poor audio quality from 3.5mm jack. Any suggestion?

XenHuang, 2022-01-25:
Ok. Just found the note saids pisignage mpv player doesn't support aux
Gapless video play (BETA)
Use of hardware accelerated mpv(https://mpv.io/) player (select through group option selection)
Audio is only through HDMI (not aux or both) as of now

So I am switching back to omxplayer, is there any configuration that I can make omxplayer to only output to the 3.5mm jack?
Can I make alias that omxplayer -o both will redirect to omxplayer -o local?

closed 2022-03-25 by XenHuang