colloqi/piSignage #168: Player2 YouTube shows web page

Flexia7th, 2022-03-09:
Just loaded up player2 and for YouTube links it shows the whole outline website with side bar and comments.

Flexia7th, 2022-03-10:
This is the link I have on the Asset
[image]
[image]

This is it loaded on the player
[image]

colloqi, 2022-03-11:
We could not load that video, can you send us the link by text or try a different video?

AbhijithBn, 2022-03-11:
We are fixing this issue and I request you to follow the steps mentioned below to play Youtube videos in fullscreen

1. Navigate to the Youtube video of your choice and Click **Share** button 
2. You will see a Share Popup. Select the Highlighted portion as shown in the image. 
[image]

3. This highlighted portion is your "Video-ID"
4. Now you can add an asset in your pisignage.com account with the following URL 

https://www.youtube.com/embed/Video-ID

make sure to add just the highlighted video-id after **embed** in the URL

example:
 Our reference video-id is **6AP8QvPkTno** hence the URL will be 
 **https://www.youtube.com/embed/6AP8QvPkTno**