colloqi/piSignage #171: [question] Is there a way to stop and switch application bettwen the "pisignage" and other program

nicolasdonbosco, 2022-03-16:
I have install the pisignage on top of the raspberry pi os (using player2/method 2)
I want to switch between my app such "chrominum" and pisignage (automatically switch to pisignage while device idle/no user interaction)
is there a way to do that?

and can I temporarily stop the pisignage? I have tried using the "pkill" command, and the piSignage automaticaly re-start again (refer to downgrade.sh script in the "shell-script" folders)
pkill -f forever
pkill -f node
pkill -f uzbl
pkill -f omx

colloqi, 2022-03-17:
You could do pm2 stop piSignage and pm2 start piSignage

hitesh83, 2022-03-21:
> 

as you mentioned chromium , you can just schedule your URL in pisigange itself , so no need to switch between apps.
for any other app as mentioned above pm2 needs to be used.

nicolasdonbosco, 2022-04-08:
thanks, pm2 stop/start is worked, but can I use original UI from raspberry OS when the app is stopped? (now when I stop the pisignage, the screen just show black screen, and some option when right click on mouse pressed) 
[image]

bontorhumala, 2022-04-08:
hi @nicolasdonbosco , did you mean after you run
`pm2 stop piSignage`

you cannot see Chromium browser / Desktop at all? I also have same problem...

nicolasdonbosco, 2022-04-08:
> hi @nicolasdonbosco , did you mean after you run `pm2 stop piSignage`
> 
> you cannot see Chromium browser / Desktop at all? I also have same problem...

yes, I can't se the chromium / desktop at all, here's the details
[image]

colloqi, 2022-04-11:
You could press F2 or ctrl+alt+T to get the terminal and issue "chromium-browser" command