colloqi/piSignage #172: Strange restart on strange hardware config

AlfioSaitta, 2022-03-18:
I'm trying to run this app on a Raspberry Pi3, connected to a video distributor (Novastar MCTRL300) on modular LED video signs.

[image]

At the first boot... and second boot... seems all work with charm...

https://user-images.githubusercontent.com/10280718/159031444-d85b4c28-96b6-4026-8059-e1fdf5b51175.mp4

After the network settings (wifi and local pisignage server), i have big issues with the screen resolution:

https://user-images.githubusercontent.com/10280718/159032164-da2bd2ef-5eff-4b09-9d68-733fbbdff214.mp4

If i use the web interface to set the video mode in "Auto(Based on hdmi EDID)" the boot process looks great, but after some seconds the device reboot alone and stall:

https://user-images.githubusercontent.com/10280718/159033158-bd32a370-4bc1-4fcb-a2df-b98e57700c63.mp4

colloqi, 2022-03-21:
Hi

1. Always server settings take over, so please use open source server UI to set "Auto" resolution
2. You can try changing hdmi_mode and hdmi_group values in /boot/config.txt to suit the resolution of your hardware
3. We have customers who use DVI interface to Novostar LED screens

hitesh83, 2022-03-21:
This issue is not related to PiSignage, it is more related to MCTRL300 and RPi config.
we are using MCTRL300 for the last 3 years now, you need to manually edit config.txt to suite.
the issue is MCTRL is having DVI input, and RPi is sending Audio on HDMI, which you convert HDMI to DVI.
MCTRL is not able to handle this part.

AlfioSaitta, 2022-03-21:
I solved the problem as you advised me... but now i should correct the resolution to that specific display:

[image]

[image]

Any advice?

hitesh83, 2022-03-21:
to get the correct display you need to use the following settings under the playlist layout,
in open source sometimes images do not fit in given dimensions, so use screen matching dimensions for images,
videos will get auto-adjusted to the below settings.
[image]

AlfioSaitta, 2022-03-21:
Finally i was able to do what i wanted. Certainly not in the best way (since i trick the css a bit to rotate the images and set the right offset), but at the moment it works for images :D

Thank you all for the tips, and i will gladly share what i have learned.

https://user-images.githubusercontent.com/10280718/159352106-d5aeb051-02a7-4a1a-8d9a-38edd0d28c90.mp4

AlfioSaitta, 2022-03-22:
Now, when run with the piSignage server, at the first boot or update, i lost the config.txt
Then i have added the working config.txt in the public folder of the server, and from the web i run

`sudo wget -O /boot/config.txt http://pi:pi@192.168.1.101:3000/config.txt; systemctl reboot`

[image]

**Can be useful to have an automatic download at the boot time of this file from the piSignage server**