colloqi/piSignage #179: Player Api /play/playlists/{file} no response

2000game, 2022-04-19:
When calling the "/play/playlists/{file}" Command the requests times out and I get not response.

`curl -X POST "http://pi:pi@10.10.1.221:8000/api/play/playlists/News" -H "accept: application/json" -H "Content-Type: application/json"`

The other commands are working just fine. News is a playlist, that is deployed on the player. I checked the error logs but there is no mention of any requests, my log level is debug. 

With /play/files/{action} I can only play files but no playlists.

AbhijithBn, 2022-04-22:
As mentioned in our API Reference Documentation https://piathome.com/apidocs/#/piPlayer/PlayOrStopAPlaylist

In order to Play a playlist, a POST request has to be sent to /api/play/playlists/<playlist-name>  along with a Body Object.
This JSON Body should have either "play" or "stop" key set to a true, depending on your requirement.

Via Postman, you can send a request this way

[image]

or using a CURL Command as mentioned below

```
curl --location --request POST 'http://pi:pi@192.168.0.106:8000/api/play/playlists/image-separate-playlist' \
--header 'Content-Type: application/json' \
--data-raw '{
    "play": true
}'
```

Please let us know if you need any other help regarding this.

2000game, 2022-04-25:
Thanks a lot, I missed that I have to send a request body.

closed 2022-04-25 by 2000game