colloqi/piSignage #181: How to return to scheduled playlist after manual change API

2000game, 2022-04-26:
When changing the playlist for players via the API, how do I put the player back into the scheduler mode? Because if I just change the playlist back to the scheduled playlist it won't auto switch to the next scheduled playlist. 
I know that I can redeploy the group via the API but I would like to know if there is a faster way. Because when redeploying the player redownloads all content.

Im working on a piece of software to play a video nearly synchronous by calling the players directly and not the managing server. After the video we play an rtmp stream. When we finished streaming we want the players to return to the scheduled content.

Also I read that I can trigger event playlists via SIGUSR2, do I have to run a script on every player that listens somehow for a request and triggers SIGUS for me or is there something already implemented?

2000game, 2023-07-22:
@rmscode 
You call it via:
`/api/play/playlists/14` 
Your request body would be:
`{"play": True}`

2000game, 2023-07-22:
Found the solution:
To return to the scheduled content call this endpoint:
`/api/play/playlists/14` 
Your request body would be:
`{"stop": True}`

14 would be the  id of the playlist that is currently playing. The player will turn black for 20-30 seconds and then will continue with the scheduled content,

closed 2023-07-22 by 2000game

rmscode, 2023-07-24:
> Found the solution: To return to the scheduled content call this endpoint: `/api/play/playlists/14` Your request body would be: `{"stop": True}`
>
> 14 would be the id of the playlist that is currently [...]

Where exactly do I find the playlist ID? I don't see it in the dashboard of the player or in the playlist JSON file.

2000game, 2023-07-24:
@rmscode 
You could try the pitv endpoint and switch off the player. If you don't have HDMI CEC properly set up the screen will just go black.

Also yes I meant the name of the playlist