colloqi/piSignage #188: Frame rate 1080p video choppy

klibansky, 2022-07-12:
Hi there, is there a specific file/codec spec that I need to use in order to get super fluid videos?
I now have a 1080p 20 second video file and I'm looking for a player that can do endless gapless looping video playback.
At the moment the video files is playing choppy and not sharp.
I'm running on a Raspberry Pi 4.

AbhijithBn, 2022-07-14:
Hi,

We do support Gapless and Endless video support on **Raspberry Pi 4** having player version **4.x.x** ( latest version being 4.7.4 ).

- VLC Media player can be used for smooth video playback.
- Once the video assets are added into a playlist, choose VLC (Experimental) under Group Settings
[image]

This will allow you to play a series of videos without any gap in loop.

klibansky, 2022-07-14:
Hi @AbhijithBn,
I tested this and still no luck.
I have a 4K monitor connected at the moment.
The video is 1080p.

I tested with VLC, which gives me a choppy playback full screen and a large gap when the video restarts.

https://user-images.githubusercontent.com/100637/178963729-5b2738eb-99b4-4e42-b1c4-372fab8ec800.mov

Then I tested with MPV with 0 second gap which gives me a 1:1 video scale (1080p) and a nicely smooth video playback but a large gap when the video restarts.

https://user-images.githubusercontent.com/100637/178963743-146009e3-7e70-4b17-908b-b734fef403f7.mov

What should the codec/settings be for the best result?

AbhijithBn, 2022-07-14:
Hi @klibansky 

We usually see a jittery video if the bitrate of the video is high. Video playback is best when bitrate is around 4-5Mbps and file format is mp4 with h264 codec.

Can you please share the source video so that I can test it at my end and give you the best solution?

klibansky, 2022-07-14:
I encoded a video in h264 4k at constant bit rate 5Mbps with vlc setting and still huge gap and very choppy video. 
At the beginning of the video you see the jump cut when the asset restarts.

https://user-images.githubusercontent.com/100637/178976701-03bf7a39-48ee-450d-b0d3-38dd825ae5ad.mov

How should I share the video?
Player version: 4.7.4/473_11_script_2022-06-11

AbhijithBn, 2022-07-18:
Hi @klibansky 
We have worked on the above issue and this will be available to use in **Release 4.7.5**

1) We have fixed VLC video player to play gapless videos by using a file-caching flag. 
2) If you have a playlist having a single video then it will be played in loop without any gaps/cuts.
3) If you have a playlist having multiple video assets, then the playlist will be played in loop with no gaps/cuts between videos.
2) However there will be a slight delay (2-3sec) in starting VLC for the first time or after a deploy is done.
3) We do not suggest MPV video player for continuous playback since it has not been optimised for Raspberry Pi yet.

klibansky, 2022-08-02:
Thats great @AbhijithBn!
I updated the player to the latest version and I can confirm that there the video is playing/looping gapless. Super!

There is still a frame rate problem however.
You can see it very clearly on the large screen. I tried to take a video of it and it looks a bit less choppy on my video but I can assure you it doesn't look smooth.

https://user-images.githubusercontent.com/100637/182364963-b92f60a2-9f08-4576-a87a-a3b22a32414b.mov

Should I use different encoding settings?

klibansky, 2022-08-24:
I'm sorry to bump this issue but I am trying to get a setup running to synchronize a number of Nfts looping on multiple screens at full resolution (4k). 1080p would be sufficient but on a 75" screen it still looks choppy. Like its for running on 30fps.

AbhijithBn, 2022-08-26:
Hi @klibansky 
Apologies for the late response.

Here are our findings upon testing.
- We noticed that the bitrate of some of the 1080p videos in your Assets library were in the range of 1Mbps - 2Mbps. This is one of the reasons for frame rate issues.
- We did not notice any frame drops in VLC Media player while testing 1080p or 4K videos.

------------------------------------------------------------------------------------------------------------------------

Here are our observations with playing videos in **VLC Media Player** on Raspberry Pi 4

1) 1080p Videos settings should be as follows
- Encoder: H264
- Format: mp4
- Bit Rate: ( 4000 to 6000 Kbps for 25-30fps videos |  ~10000 Kbps for 60fps videos ) 
- Frame Rate: 25, 30, 60 fps

2) 4K Videos settings should be as follows:
- Encoder: H265 (also known as HEVC)
- Format: mkv
- Frame Rates: 30 or 60fps

------------------------------------------------------------------------------------------------------------------------

Note:
- We currently do not have server support for 4K videos and hence when you upload a 4K video, it gets converted to a 1080p video. We are working on the server 4K support.
- We request you to upload 4K videos as per above mentioned settings directly to player using the Player-UI ( ie: http://<player-ip-address>:8000) and test 4K Videos using the "Play" button or by adding the video to a playlist.
- Here are some sample 4K mkv format videos for you to test out at your end
https://4kmedia.org/sony-beautiful-sea-in-okinawa-uhd-4k-demo/ (30fps video)   
https://4kmedia.org/sony-bravia-uhd-hdr-4k-demo/   (60fps video)

Samuel12321, 2023-05-27:
Is server support for 4k videos resolved yet?