colloqi/piSignage #189: Player unable to boot up with error black screen

ptc-dev, 2022-08-24:
Pi 4 model B 2GB has issue with unable to boot to OS.

[image]

Pi 4 model B 2GB has issue with unable to boot to OS.

[image]777777777777777777777777777777777777777777777777777

Need to restart several time to get it boot in correctly.

Anyone has a solution?
Need to restart several time to get it boot in correctly.

Anyone has a solution?

hitesh83, 2022-08-24:
Haven't got such issue till date, error on screen says unable to open sdcard, so will suggest you to try with fresh sd card.

ptc-dev, 2022-08-24:
we are experience multiple pi player with this issue on Raspberry pi 4
All SD card are new.

SD Card:
Kingston Canvas Select Plus 32GB 100MB/s

hitesh83, 2022-08-24:
The pi4 boot loader is not able to read the SD card, that's the error on your screen.
so the issue is before the PiSignage image starts. 
You may try another version of Image or with a fresh sd card, that's the only solution for this.

ptc-dev, 2022-09-09:
we have been facing the unable to read SD card with Raspberry pi 4.

we do not have any issue with Raspberry pi 3. This issue only happen in Raspberry pi 4, we have tried version 4.7.3 and 3.2.0.

could you please suggest what version will be the most stable for Raspberry pi 4b 2 GB for Player?

colloqi, 2022-09-12:
Both should work, 3.2.0 is preferred as of now.