colloqi/piSignage #196: New web app / Player 2

Narrowcastingscherm, 2022-11-17:
I am trying out the new web app / player 2 website, however i cannot connect to my open source server with this option.

AbhijithBn, 2022-12-15:
With the Web Player, you won't be able to connect the player to an open Source server.
To solve this issue, you could use our Android App which works on Android Mobile and Android TV. 
Link to app: https://play.google.com/store/apps/details?id=com.pisignage.player2

hitesh83, 2022-12-22:
both are not working with the open source server updated Nov 24 2022

AbhijithBn, 2022-12-27:
We will soon release an update to the android app which should solve the issue.

AbhijithBn, 2023-01-01:
We have rolled out a new release for our android application. (version 4.8.0)
Can you check if open source is working now @hitesh83  @Narrowcastingscherm?

hitesh83, 2023-01-02:
works well now with HTTP on the 3000 port of open source.
I tested jpg, and mp4 both work well including a screenshot.
The webpage won't load fully and the address bar is visible including the bottom translation option. No screenshot for webURL.

AbhijithBn, 2023-01-02:
Thank you @hitesh83.
Yes, there are some limitations in opening a webpage with CORS support
1) Not in fullscreen
2) URL visible at the top address bar
3) Screenshot cant be captured
4) Only works for main zone assets
5) Webpage (Cors support) not available on Android TV (only available on mobile)

hitesh83, 2023-01-03:
@AbhijithBn Also we noticed that the playlist gets stuck on the webpage, after closing the app and reopening it works normally.