colloqi/piSignage #201: CEC commands don't always work on first attempt

stpeteraz, 2022-12-31:
Hello,

There are certain TVs that require multiple attempts via CEC commands to turn them on and off. My recommendation is to issue these commands multiple times in case the TV didn't register the first time along with setting as the active source.

The TVs we're having issues with are smart fire TVs that turn on, but linger around on the home screen, as if they don't know which input to switch to. I've done some testing and the TVs had a 100% success rate if I ran the command multiple times with a 3 second delay between each one. I essentially tried to mimic what was occurring by going into the Pi shell and attempting to turn these TVs on while they were in a standby state:
[image]

First instance failed, I ran another time, and the CEC client reported the same thing, a third time, and it was able to turn the TV on. I believe there should be method in the code where it pushes out a certain amount of commands with potentially adjustable delays (can be adjusted in group settings or piSignage server settings)?

wrldtvlr, 2022-12-31:
I've suggested this as well and strongly support this. We have multiple new LG Smart TVs that exhibit this behavior.

Samuel12321, 2023-05-21:
Also run into this issue.

AbhijithBn, 2023-05-21:
Hi @Samuel12321 , can you share the player forever_out and forever_err.log with us at support at pisignage.com?

Samuel12321, 2023-05-21:
Im away for a few days, but will do when i get back.
However i have noticed that the CEC off command in particular doesnt seem to reliably work with LG and Samsung TV's

Samuel12321, 2023-08-30:
@AbhijithBn Has there been any progress with this?

AbhijithBn, 2023-08-30:
@Samuel12321 
As discussed over support, from the logs that you shared
 
2023-06-22T17:37:57: cec support: undefined; stderr: exec error: Error: Command failed: echo scan | cec-client -s -d 1
2023-06-22T17:37:57:
2023-06-22T17:37:57: *** CEC is NOT supported ***
 
You can try issuing the following command to check the status of CEC (you should see 2 devices)
 
`echo scan | cec-client -s -d 1
`