colloqi/piSignage #205: Android app is not showing section containing link/iframe

psantos9, 2023-02-02:
I am running a playlist in a group composed by a Raspberry Pi and an Android device.
The playlist has a 2-zone layout, and on the bottom zone I've set it to show a link.

The Raspberry PI device is showing the bottom link:
[image]

The Android device is not showing the link:
[image]

Is this a limitation from the android app, or am I missing something here?

Thank you

psantos9, 2023-02-06:
So I've investigated a bit on this issue and here is more info on it:

**For testing I've defined a three-zone custom layout defined as follows:**
[image]

While inspecting the dom of the player2 app there are a detail that immediately poppep up to my eye. There are two div elements with the same id =_#main_.

[image]

Android app only shows the _#main_ container:
[image]

Player2 app on Chrome shows the correct layout:
[image]

Any clues here? Thank you

AbhijithBn, 2023-02-07:
Hi @psantos9 

If you navigate to Player Settings by clicking on the D_PAD Center button, you can see the Player Resolution.
Example: 960x540
[image]

You have to design your custom layout according to this dimension and deploy it to the player. 
Let us know if this worked.

psantos9, 2023-02-07:
Hi @AbhijithBn,
I'm seeing that same 960x540 resolution on my android player settings, but the TV is HD (1280 x 720).
Where is this 960x540 value coming from?
Thank you

AbhijithBn, 2023-02-07:
In Android Player, Group Settings cannot be used to control Display Resolution, Orientation, and Multi Screen layout.

Android devices not only have [different screen sizes](https://developer.android.com/training/multiscreen/screensizes)—handsets, tablets, TVs, etc.—but also have screens with different pixel sizes.
Reference: https://developer.android.com/training/multiscreen/screendensities

psantos9, 2023-02-07:
Got it. Thank you!

closed 2023-02-07 by psantos9