colloqi/piSignage #209: Playlist not showing available assets

godsgood33, 2023-03-10:
Web app does not show the uploaded available assets on the playlist screen.  I have tried reloading the SD card and it starts out working, but then stops.  Also, it seems that after I look at the playlist the first time it stops showing any playlists in the list and the web server crashes.  It's at the latest version 3.5.2.

[image]

[image]

[image]

forever_err.log output
```
TypeError: Cannot read property 'find' of undefined
    at /home/pi/piSignagePro/pi-server.js:7:30292
    at nextTask (/home/pi/piSignagePro/node_modules/async/dist/async.js:5756:27)
    at next (/home/pi/piSignagePro/node_modules/async/dist/async.js:5764:13)
    at /home/pi/piSignagePro/node_modules/async/dist/async.js:324:20
    at /home/pi/piSignagePro/pi-server.js:7:30207
    at FSReqCallback.oncomplete (fs.js:171:23)
connect.limit() will be removed in connect 3.0
connect.multipart() will be removed in connect 3.0
visit https://github.com/senchalabs/connect/wiki/Connect-3.0 for alternatives
connect.limit() will be removed in connect 3.0
(node:915) [DEP0066] DeprecationWarning: OutgoingMessage.prototype._headers is deprecated
(Use `node --trace-deprecation ...` to show where the warning was created)
(node:915) [DEP0106] DeprecationWarning: crypto.createDecipher is deprecated.
connect.limit() will be removed in connect 3.0
connect.multipart() will be removed in connect 3.0
visit https://github.com/senchalabs/connect/wiki/Connect-3.0 for alternatives
connect.limit() will be removed in connect 3.0
(node:859) [DEP0066] DeprecationWarning: OutgoingMessage.prototype._headers is deprecated
(Use `node --trace-deprecation ...` to show where the warning was created)
(node:859) [DEP0106] DeprecationWarning: crypto.createDecipher is deprecated.
/home/pi/piSignagePro/pi-server.js:7
}}),c.register("\x2e\x2f\x61\x70\x70\x2f\x63\x6f\x6e\x74\x72\x6f\x6c\x6c\x65\x72\x73\x2f\x70\x69\x2d\x65\x76\x65\x6e\x74\x73\x2e\x6a\x73",function(a,b,c){"use strict";var m,n,d=c("\x66\x73"),e=c("\x70\x61\x74\x68"),f=c("\x61\x73\x79\x6e$                                                                                                                                                                                                                                            $
TypeError: Cannot read property 'find' of undefined
    at /home/pi/piSignagePro/pi-server.js:7:30292
    at nextTask (/home/pi/piSignagePro/node_modules/async/dist/async.js:5756:27)
    at next (/home/pi/piSignagePro/node_modules/async/dist/async.js:5764:13)
    at /home/pi/piSignagePro/node_modules/async/dist/async.js:324:20
    at /home/pi/piSignagePro/pi-server.js:7:30207
    at FSReqCallback.oncomplete (fs.js:171:23)
connect.limit() will be removed in connect 3.0
connect.multipart() will be removed in connect 3.0
visit https://github.com/senchalabs/connect/wiki/Connect-3.0 for alternatives
connect.limit() will be removed in connect 3.0
(node:864) [DEP0066] DeprecationWarning: OutgoingMessage.prototype._headers is deprecated
(Use `node --trace-deprecation ...` to show where the warning was created)
(node:864) [DEP0106] DeprecationWarning: crypto.createDecipher is deprecated.
```

zazeejesal, 2023-04-14:
having the same issue on 3.2.6, crashes after viewing playlist locally

nrayburn-14G, 2023-11-30:
I'm having the same issue as well on version 3.2.8. It also crashes after viewing playlists locally.

W0rkingChr1s, 2024-05-06:
Still no solution for this?
Heading to the same prob.