colloqi/piSignage #214: able to boot up

ptc-dev, 2023-06-23:
unable to boot into system with error boot up screen. I have attched a video to show the error.

https://github.com/colloqi/piSignage/assets/105030123/168e422d-1262-4059-8fd7-b0e14cb00009

colloqi, 2023-06-23:
I guess the SD card is corrupted, please reprogram/change the SD card

ptc-dev, 2023-06-23:
i have tried multiple SD card with new burn ISO file. Still having the same problem

stormvansoldt, 2023-07-13:
What program are you using to write the ISO to the SD card? If this is happening to multiple cards then it must be the writing software causing these issues.

colloqi, 2023-07-14:
https://etcher.balena.io/ or Raspberry Pi imager, also make sure to use 2.9.7 if the hardware is Pi Zero,1,2

ptc-dev, 2023-07-19:
Hi Ravi,

We keep experiencing error or stable player for pi4.
It often come with unable to boot screen.

we have tried multiple burning software such as balena Etcher, Raspberry pi imager, windows iso burner ,tried with different brand sd card and different power supply.

[A computer screen with a qr code on it  Description automatically generated]

Could you please advise solution for the issue?

Kind regards

________________________________
Peter Chai l Project & IT Manager Assistant
Address: Unit 3, 116 Magnesium Drive, Crestmead 4132 QLD
Mobile: +61 430 213 232
Tel: +61 738 008 189
Email: ***@***.***
***@***.***
NOTICE OF CONFIDENTIALITY: The content of this email is confidential and intended for the recipient(s) specified in message only. It is strictly forbidden to share any part of this message with any third party, without a written consent of the sender. If you received this message by mistake, please reply to this message and follow with its deletion, so that we can ensure such a mistake does not occur in the future.

colloqi, 2023-07-27:
1. Which link you are downloading?
2. is it MAC or windows?
3. Raspberry Pi imager works very well (so as other programs)
4. can you send a screenshot after powering ON?