colloqi/piSignage #218: Bug: dhcpcd.service fails to start after "Factory reset" in the settings page of player WebUI

rmscode, 2023-07-26:
This is an odd one - PiSignage 3.2.0

We have a webserver that is hosting a webpage we intended on using as an "asset" for PiSignage to display. Access to the URL on our webserver is restricted to whitelisted IP addresses only.

The pi, with its DHCP assigned IP, accesses the URL as expected with a freshly configured instance of PiSignage. However, after performing a "factory reset" in PiSignage, the Pi is no longer able to access the URL unless I whitelist our firewall's IP. The pi and firewall are on the same subnet. The Pi is also hardwired, not on WiFi.

Anyway, I was able to reproduce this 3 times. Every time I "factory reset" a fresh install of the PiSignage . . . the whitelist doesn't work unless the firewall IP is also whitelisted. I noticed after each factory reset/reboot the message "Failed to start dhcpcd.service on all interfaces" during boot.

After messing around quite a bit I think I figured out what is happening. PiSignage is writing to /etc/network/interfaces after a "factory reset" is performed. I think that file is used by dhclient (network.services) . . . and that is where the conflict is?

I don't forecast using the factory reset option at all, especially with this issue, but it would still be nice to know how to reverse the changes in the event it was used...instead of reinstalling the OS each time.

I've noticed that dhcpcd and dhclient are both running on this PiSignage Raspberry OS img after a fresh install. I don't know if that's typical, but I'm assuming that it's not. 

I've tried deleting the /etc/network/interfaces file and disabling dhclient with no luck . . . still have the strange issue with the web server seeing the Pi present itself with the firewalls IP.

Should have mentioned this earlier, but the Pi is still accessible via its initial dhcp assigned IP after a PiSignage "factory reset". 

**Pre "factory reset" on a fresh instance of PiSignage - dhcpcd.service:**
```
pi@raspberrypi:/var/lib/dhcpcd5$ systemctl status dhcpcd.service
* dhcpcd.service - dhcpcd on all interfaces
   Loaded: loaded (/lib/systemd/system/dhcpcd.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/dhcpcd.service.d
           `-wait.conf
   Active: active (running) since Wed 2023-07-26 20:32:16 IST; 3min 47s ago
  Process: 379 ExecStart=/usr/lib/dhcpcd5/dhcpcd -q -w (code=exited, status=0/SUCCESS)
 Main PID: 686 (dhcpcd)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/dhcpcd.service
           |-494 wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0 -Dnl80211,wext
           `-686 /sbin/dhcpcd -q -w

Jul 26 20:32:16 raspberrypi dhcpcd[379]: forked to background, child pid 686
Jul 26 20:32:16 raspberrypi systemd[1]: Started dhcpcd on all interfaces.
Jul 26 20:32:16 raspberrypi dhcpcd[686]: wlan0: carrier acquired
Jul 26 20:32:16 raspberrypi dhcpcd[686]: wlan0: IAID dd:15:63:00
Jul 26 20:32:16 raspberrypi dhcpcd[686]: wlan0: adding address fe80::baac:d769:ad01:d538
Jul 26 20:32:16 raspberrypi dhcpcd[686]: wlan0: probing address 192.168.100.1/24
Jul 26 20:32:16 raspberrypi dhcpcd[686]: wlan0: soliciting an IPv6 router
Jul 26 20:32:21 raspberrypi dhcpcd[686]: wlan0: using static address 192.168.100.1/24
Jul 26 20:32:21 raspberrypi dhcpcd[686]: wlan0: adding route to 192.168.100.0/24
Jul 26 20:32:45 raspberrypi dhcpcd[686]: wlan0: no IPv6 Routers available
pi@raspberrypi:/var/lib/dhcpcd5$
```

**Pre "factory reset" on a fresh instance of PiSignage - /etc/network/interfaces:**
```
# interfaces(5) file used by ifup(8) and ifdown(8)

# Please note that this file is written to be used with dhcpcd
# For static IP, consult /etc/dhcpcd.conf and 'man dhcpcd.conf'

# Include files from /etc/network/interfaces.d:
source-directory /etc/network/interfaces.d
```

**The dhcpcd.service status after a "factory reset":**
```
pi@testing:/var/lib/dhcpcd5$ systemctl status dhcpcd.service
Warning: The unit file, source configuration file or drop-ins of dhcpcd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
* dhcpcd.service - dhcpcd on all interfaces
   Loaded: loaded (/lib/systemd/system/dhcpcd.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/dhcpcd.service.d
           `-wait.conf
   Active: failed (Result: exit-code) since Wed 2023-07-26 19:49:57 IST; 29min ago
  Process: 437 ExecStart=/usr/lib/dhcpcd5/dhcpcd -q -w (code=exited, status=6)

Jul 26 19:49:57 testing systemd[1]: Starting dhcpcd on all interfaces...
Jul 26 19:49:57 testing dhcpcd[437]: Not running dhcpcd because /etc/network/interfaces
Jul 26 19:49:57 testing dhcpcd[437]: defines some interfaces that will use a
Jul 26 19:49:57 testing dhcpcd[437]: DHCP client or static address
Jul 26 19:49:57 testing systemd[1]: dhcpcd.service: Control process exited, code=exited, status=6/NOTCONFIGURED
Jul 26 19:49:57 testing systemd[1]: dhcpcd.service: Failed with result 'exit-code'.
Jul 26 19:49:57 testing systemd[1]: Failed to start dhcpcd on all interfaces.
```

**The /etc/network/interfaces after a "factory reset":**
```
auto eth0
auto lo
allow-hotplug eth0
allow-hotplug usb0
allow-hotplug usb1

iface lo inet loopback

iface eth0 inet dhcp

iface usb0 inet static
      address 192.168.42.10
      netmask 255.255.255.0
      gateway 192.168.42.129
iface usb1 inet static
      address 192.168.42.10
      netmask 255.255.255.0
      gateway 192.168.42.129

allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
wireless-power off
iface default inet dhcp
```

colloqi, 2023-07-29:
Can you avoid factory reset and instead remove files /home/pi/piSignagePro/config/_settings.json and _config.json files? It has the same effect as factory reset.

rmscode, 2023-07-29:
I will test this.

rmscode, 2023-07-31:
@colloqi @ravibail l I have tested this for you . . .

- Installed fresh copy of 3.2.0
  - Completed basic configuration of player. Created a playlist and asset. Everything is working as expected.
- Removed `_settings.json` and `_config.json` via SSH
- `sudo reboot now`
- Expected result achieved
  - Settings have been reset.
  - `etc/network/interfaces` file remains unmodified
  - dhcpcd starts properly
  - Player displays content

rmscode, 2023-08-14:
@ravibail @colloqi 

I have just discovered that updating "Advanced Users - Edit Settings File" causes the same issue!! I made an edit to the overscan settings to be specific.
[image]