colloqi/piSignage #219: Regarding the reboot option in the _settings.json file (3.2.0)

rmscode, 2023-07-31:
I've noticed the following in the _settings.json file:

```
"reboot": {
        "enable": false,
        "time": null
    }
```

However, changing the values do not appear to reboot the player. I'm wondering what this is for.

AbhijithBn, 2023-08-01:
@rmscode  Can you try 
```
"reboot": {
    "enable": true,
    "time": "1970-01-01T03:30:00.000Z",
    "absoluteTime": "09:00"
}
```

Please note that, if your player is connected to your server, then the server settings will override the player settings.

AbhijithBn, 2023-08-16:
Can you check if a CronJob is created in your Raspberry Pi by checking the output of the following command?
`crontab -l`

rmscode, 2023-08-16:
> 
Ah . . . so that setting creates a cron job. I can confirm that a cron job has been created for root - `/sbin/reboot && echo piSignage`. It should work if a cron job exists.. Maybe in I missed something during my original testing. I'll try again. Thanks @AbhijithBn

rmscode, 2023-08-16:
I understand why I believed that it didn't work in my testing.

I assumed that `absoluteTime` was the key that needed to be changed, but its actually `time`.

Future readers, update the time between the angle brackets . . . Don't leave them in the json code though: 
```
"reboot": {
        "enable": false,
        "absoluteTime": "1970-01-01T<00:00>:00.000Z", 
        "time": "09:00"
    }
```

closed 2023-08-16 by rmscode