colloqi/piSignage #220: 3.2.0 to 3.2.8 Upgrade Failing

baaron4, 2023-08-07:
We have an issue on one of our piSignage players set to layout **2C: Two equal size zones.** where chromium seems to fail with Error:5. Then the applications gets restarted. 

It doesnt look like another pi we have running that same dashboard is having this issue and it is running 3.2.8. So I am trying to get this one upgraded to 3.2.8 using the update button in the server.

After I clock the update button I can see the player performing an update, it reboots, it trys to display a screen and eventually goes back to Error Code: 5. Looking at the server it still reads 3.2.0.

How can I manually update this player?

rmscode, 2023-08-07:
Take a look at this #217.

colloqi, 2023-08-09:
Also try 
`sudo apt update;  sudo apt -y full-upgrade &`

to upgrade the OS

baaron4, 2023-08-09:
@rmscode I was able to look at thread 217 to get an image file containing the 3.2.8 image V14 (after a node check). I added it to the pi and ran the update command however it still failed to update.

@colloqi The first half of that command ran fine but the second half is an invalid operation:
[image]
I went ahead and tried to update the image via the server again (and this time it did do a stop action for about 1.5 minutes rebooted etc) and it looked like it applied an update. However the server still reports 
[image]
and its still crashing chromium with error 5
[image]

The one that is working is: 
[image]

AbhijithBn, 2023-08-16:
Hey @baaron4 

The second command is supposed to be 
`sudo apt -y full-upgrade` 

There is a typo at "upgrade" -> uprade

Once you have upgraded your player, reboot the player by issuing 
`sudo reboot`