colloqi/piSignage #221: Access to Chromium Dev Tool

rmscode, 2023-09-06:
I have an issue loading a custom webpage that only happens on my PiSignage 3.2.0 players. I need access to the chromium dev tools so I can investigate . . . or I need a way to directly update the instance of Chromium the player uses.

colloqi, 2023-09-07:
You could ssh to Pi or press ctrl+alt+T to get into the shell. piSignage is based on Raspberry OS (3.2.x is legacy OS). Then you can update the chromium.

rmscode, 2023-09-07:
> You could ssh to Pi or press ctrl+alt+T to get into the shell. piSignage is based on Raspberry OS (3.2.x is legacy OS). Then you can update the chromium.
Yes, I am aware of that. The support docs on your website say that the best way to update Chromium is to update the PiSignage image. Also, Chromium isn't listed when you use the command `apt list --upgradable`. Those two facts lead me to make this post. 

I just really want to be able to access dev tools in Chromium.