colloqi/piSignage #222: Node v14 deprecated

Havbaba, 2023-09-11:
Why is still Node.js v14 used, it's giving error's all over the place.
Including timers/promises not being found and when googling this it only starts from Node.js 15?!

AbhijithBn, 2023-09-12:
Hey @Havbaba 

Can you let us know if you are having issues in the open-source server or with the player?
We would also need more details about the error to help you.

Havbaba, 2023-09-12:
Hi,

I am using an Orange Pi 5B and trying to install the Player 2 which should support Debian 10/Debian 11/Ubuntu 20.
Tried:
- Orangepi5b_1.0.2_ubuntu_jammy_desktop_gnome_linux5.10.110
- Orangepi5b_1.0.2_debian_bullseye_desktop_kde-plasma_linux5.10.110
- Armbian_23.8.1_Orangepi5_jammy_legacy_5.10.160
- Armbian_23.8.1_Orangepi5_bookworm_legacy_5.10.160_cinnamon_desktop
Following used for the images:
- Orangepi5b_1.0.2_ubuntu_focal_desktop_xfce_linux5.10.110

It all results to the same:
Node.js 14 deprecated warnings, I just let the script do it's thing.
I don't know if I am supposed to update them myself or just leave it.

When the script finishes I do sudo reboot and it boots to the pi user desktop.
I try to run start.sh, resulting in following error: [https://ibb.co/N2S9bS0](https://ibb.co/N2S9bS0)
Error [ERR_MODULE_NOT_FOUND]: Cannot find package 'pm2' imported from /home/pi/player2/monitor.mjs

AbhijithBn, 2023-09-12:
Hey,

Did you get any other errors during the installation process of Player2 while following the instructions mentioned at https://pisignage.com/releases/Player2_installation_procedure.html?

Can you check the versions of npm and node by issuing 
```
node -v
npm -v
```

Can you navigate to /home/pi/player2 and issue the following command?
```
sudo npm install pm2 -g
npm install

```

This should install all the player dependencies. Once that is done, reboot your player and check if you have errors.

Havbaba, 2023-09-12:
Hi,

Thanks for your fast reply and for helping!

So first it started with the deprecated error: [https://ibb.co/D1N9fvQ](https://ibb.co/D1N9fvQ)

And a bit further it has these:
[https://ibb.co/VLKJhD2](https://ibb.co/VLKJhD2)
[https://ibb.co/x2kDCmG](https://ibb.co/x2kDCmG)
This timers/promises error shows up 3 times next to each other.

Node
v14.21.3

When doing the npm -v I get the timers/promises error as shown in the pictures:
Error: Cannot find module 'timers/promises'

Shall I proceed with the install of pm2 or shall we fix the npm error first?

AbhijithBn, 2023-09-12:
Can you update npm?

`sudo npm i update npm -g
`

Havbaba, 2023-09-12:
No, gives the same Error: Cannot find module 'timers/promises'

AbhijithBn, 2023-09-12:
According to the errors logs that you have shared, npm version is 10.1.0 and Node14 is compatible with with npm version 6.x.x

Can you try these steps

1) Uninstall node by issuing `sudo apt-get purge nodejs`
2) Check if npm and node still return any versions by issuing `node -v` and `npm -v`
3) Setup source list for Node14 by issuing `curl -sL https://deb.nodesource.com/setup_14.x | sudo bash -`
4) Install node using `sudo apt-get install -y nodejs`

The errors are due to Node14 not being compatible with npm10

AbhijithBn, 2023-09-12:
Here is your Reference to compatibility issue according to your logs 
[image]

Havbaba, 2023-09-12:
Aha okay, interesting!

Now:
Node
v14.21.3

NPM
6.14.18

Did a reboot and started up to desktop again, tried sh start.sh
Gave the pm2 error again and `sudo npm install pm2 -g`
Which leads to following new error, progress!
TypeError: isexe is not a function
[https://ibb.co/YkwPQKL](https://ibb.co/YkwPQKL)

AbhijithBn, 2023-09-12:
Please reach out to us at support at pisignage.com and we will guide you.

Havbaba, 2023-09-12:
https://help.pisignage.com/hc/en-us/community/posts/22898867713177-OrangePi-5B-installation-issues-with-npm