colloqi/piSignage #224: Long Blackscreen on transition to any Webpage

CristianBuschi, 2023-10-04:
My PiSignage software runs on an Raspberry Pi 3 B 
Software version: 3.2.8
Browser: Chromium
Transitioning from any asset type to any webpage leads to a long loading time (>5s) and black screen
And its not only on the first load - everytime you transition to a webpage (even if it is the same page) you have a long loading time with a black screen.
Transitioning to other assets like images or videos is almost instant

Has anyone experienced similar issues or does anyone how to fix (or at least improve) this?

Thank you in advance for helping me out :)

colloqi, 2023-10-08:
You could try "keep webpages in memory" option under Group settings

CristianBuschi, 2023-12-15:
Sorry for the late awnser
This solved my issue 👍 

Thank you

closed 2023-12-15 by CristianBuschi