colloqi/piSignage #225: player2 installation not work with the latest raspberry OS (bookworm version) / node.js 14 is depracated

pillgrim, 2023-11-13:
I have tried a new installation with the latest raspberry os version (Debian version: 12 (bookworm)). 
After a restart, raspberry pi doesnt startup. it stops with a blinking underline.

additionally, nodejs version 14 is depracted an i receive the message (see attachment)
[image]

colloqi, 2023-11-21:
Hi, We are working on bookworm release and planned to be released before Nov 23 end

rmscode, 2023-12-08:
> Hi, We are working on bookworm release and planned to be released before Nov 23 end

How is the bookworm release coming along?

christian-andersson, 2023-12-14:
Any status update for this? currently this is stopping us from deploying the player on multiple raspberry pi:s

Or where do I need to look and what do I need to look into the find out what is failing.

hitesh83, 2023-12-14:
> Any status update for this? currently this is stopping us from deploying the player on multiple raspberry pi:s
> 
> Or where do I need to look and what do I need to look into the find out what is failing.

Did your installation on player2 go fine on pi5 ?

christian-andersson, 2023-12-14:
> > Any status update for this? currently this is stopping us from deploying the player on multiple raspberry pi:s
> > Or where do I need to look and what do I need to look into the find out what is failing.
> 
> Did your installation of player2 go fine on pi5 ?

I have only tested it with the 64 and 32 bit version of Raspbian (Bookworm) downloaded from https://www.raspberrypi.com/software/operating-systems/ on a Raspberry pi 4.

got the same warnings, and the same result, a black screen wherethere is just a blinking cursor (nothing to to stop, etc)

AbhijithBn, 2023-12-14:
The current 4.9.3 image will not work on Raspberry Pi 5.

We are implementing piSignage on Raspberry Pi 5 and it will be out very soon. Will keep you updated as we release the same. 
Thanks for your patience in this regard and thank you for supporting PiSignage!

christian-andersson, 2023-12-14:
I am trying to follow this description on raspbian bookworm in a raspberry pi 4, not pi 5
https://pisignage.com/releases/Player2_installation_procedure.html

AbhijithBn, 2023-12-15:
Please use the 4.9.3 image available at https://github.com/colloqi/piSignage#getting-the-player-ready to setup piSignage for Raspberry Pi4.

rmscode, 2023-12-15:
> Please use the 4.9.3 image available at https://github.com/colloqi/piSignage#getting-the-player-ready to setup piSignage for Raspberry Pi4.

The "player2" links provided for the RPi 4 are for version 4.9.0.

rmscode, 2023-12-20:
@christian-andersson The bookworm release is ready! https://pisignage.com/releases/Player2_installation_procedure.html

rmscode, 2023-12-20:
@colloqi @AbhijithBn @christian-andersson I did some _very basic_ testing (displayed a single web page running a WASM app) this morning on both a raspberry pi 3 and raspberry pi 4. PiSignage + Bookworm works on both...very minor issue on the Pi 3 however.

I initially prepared the SD card and installed PiSignage on the Raspberry Pi 4. PiSignage worked as expected and would start automatically each time the Pi was rebooted. I used the same SD card to boot up the raspberry pi 3 and PiSignage needed to be started manually via the `start.sh` script. I'm not sure why that is. It works well otherwise.

colloqi, 2023-12-21:
This image is based on wayland and is available only on Pi 4 & Pi 5 now as per https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/, 

` The performance of Wayfire on earlier platforms is still being optimised, so for now they will continue to run the old X11 display server and the Openbox window manager, but at some point these platforms will also be switched to Wayfire.)`

so till then we recommend 3.2.0 for Pi 3 and below.