colloqi/piSignage #226: Some internet sites not displaying

AleksanderKruecken, 2023-11-24:
Hi, 

I have Raspberry Pi 3 (1 GB) and Pisignage 3.2.0 (Raspbian buster) installed on it. 
I have noticed that some Web links (Assets - > Add - > Add a link - > Web link) do not get displayed.
For example https://github.com
Some other works, for example https://ubuntu.com

Forever_out log:
Browser stderr: Fontconfig error: Cannot load default config file: No such file: (null)

I have done apt update and upgrade, apt install fontconfig, etc. but nothing helps.

pi@ArtesWall1:/etc/fonts/conf.avail$ ls
pi@ArtesWall1:/etc/fonts/conf.d$ ls
return:
20-unhint-small-dejavu-lgc-sans-mono.conf 20-unhint-small-dejavu-sans-mono.conf 30-droid-noto-mono.conf 57-dejavu-serif.conf 58-dejavu-lgc-serif.conf
20-unhint-small-dejavu-lgc-sans.conf 20-unhint-small-dejavu-sans.conf 57-dejavu-sans-mono.conf 58-dejavu-lgc-sans-mono.conf 65-droid-sans-fallback.conf
20-unhint-small-dejavu-lgc-serif.conf 20-unhint-small-dejavu-serif.conf 57-dejavu-sans.conf 58-dejavu-lgc-sans.conf

fonts.conf doesn't exist in /etc/fonts
I have manually created fonts.conf and put the following content in it.
https://www.stata.com/support/faqs/unix/fonts.conf/lux64/fonts.conf

A have also tried:
export FONTCONFIG_PATH=/etc/fonts

Nothing of above works and some webpages still don't get displayed.
What could be the problem?

Kind regards.
[image]
[image]

AbhijithBn, 2023-12-01:
Hi @AleksanderKruecken 

Can you try adding the link as "Web Page (opens a new browser for CORS support)"

[image]

AleksanderKruecken, 2023-12-01:
@AbhijithBn Thanks, your suggestion worked.

closed 2023-12-01 by AleksanderKruecken