colloqi/piSignage #231: PiSignage Player-2 and open source Server

CristianBuschi, 2023-12-15:
Hey all,

the https://pisignage.com/player2/ does not connect to my open source server.
I found here: https://github.com/colloqi/piSignage/issues/196 that it is not currently supported.
Or at least it wasnt supported in Nov, 2022.
Is it still that way?
Or is there a way to connect player2 to an open source server?

AbhijithBn, 2023-12-18:
Hi @CristianBuschi 

The progressive web app (PWA) at https://pisignage.com/player2 needs a https connection to the server.

The open-source server should be secured using SSL certification in order to be used with https.
 
The best way is to use nginx as reverse proxy to re-route traffic from https to port 3000,  https://docs.nginx.com/nginx/admin-guide/web-server/reverse-proxy/

CristianBuschi, 2024-01-24:
I tested nginx as reverse proxy and it worked.
Since PWA uses a Websocket connection this also helped me to configure nginx:
https://www.nginx.com/blog/websocket-nginx/

And I also want to thank you for your response