colloqi/piSignage #234: Cannot create horizontal + vertical layout with dual display

controlol, 2024-01-02:
Platform: RPi 4
Player version: 4.9.5/490_11_script_2023-04-23

When creating a custom layout for dual displays the *side* container cannot be modified. Same is true for the *main* container.

When loading a *horizontal custom layout* both displays are treated as horizontal, even though one display is configured as vertical using the template designer. Adding custom CSS does not have an effect either. 

This was the CSS I added.

```css
#side {
  transform: rotate(90deg) !important;
  -webkit-transform: rotate(90deg) !important;
}
```

[image]

Perhaps the template designer was never meant for this, perhaps it is a bug.

I think it would be very useful to have separate playlists/scheduling for each display. If this were possible the layout could be a bonus for each display instead of a requirement, and losing the functionality to control the layout of the display.

AbhijithBn, 2024-01-03:
Hi

piSignage for Raspberry Pi connected to dual display works when both the displays are used in Landscape (horizontal) or Portrait (vertical) mode.

Please make use of Layout 2a/ 2b/ 2c rather than a custom-layout and assign a playlist with a Portait mode to the side zone. 
[image]

Here is a support article for your reference about certain limitations.
https://help.pisignage.com/hc/en-us/articles/21655587985433-Dual-Display-with-PiSignage

controlol, 2024-01-04:
Okay, thank you for verifying. I think we will be using 2 pi's for now.

closed 2024-01-04 by controlol