colloqi/piSignage #235: Mouse cursor stuck on screen

Noddley, 2024-01-03:
No matter what commands I run, I can’t get the mouse cursor off the screen! I can connect a mouse and move it but the second it reboots the mouse is centered on the screen. I have a 8gb ram with 256gb micro sd card raspberry pi 4. Please help!!

colloqi, 2024-01-04:
Hi, can you upgrade to 5.0.3 and check, we have fixed this issue in the same

Noddley, 2024-01-04:
Will try it! I’ll keep you posted

Noddley, 2024-01-04:
So when I try to upgrade the pi's from the open score sever portal it updates half way then restarts and in the logs it give me a error saying "socket is not available" now looking at it. On the system start screen for the pi's it says it's disconnected from the sever but I can clearly push files to it and it's showing up on my server. Any help would be appreciated

Noddley, 2024-01-04:
[forever_out (1).log](https://github.com/colloqi/piSignage/files/13836263/forever_out.1.log)
Here is the log

AbhijithBn, 2024-01-05:
Hey @Noddley 

Can you please send the forever.log instead of the forever_out.log
**forever.log** has the upgrade logs

Noddley, 2024-01-05:
I see that, my bad here you go;
[forever (1).log](https://github.com/colloqi/piSignage/files/13843022/forever.1.log)

AbhijithBn, 2024-01-08:
Hi @Noddley 
http://10.0.0.230:3000/releases/piimage5.0.3-p2-v20.zip
has a 404 response code. Can you please update your open source server using "git pull origin master" command?

Noddley, 2024-01-09:
I will try to delete json file and the nodes because if I’m not mistaken if it’s on docker compose it updates every time it’s restarted

jacobmessner, 2024-01-09:
Hello,

I am experiencing the same issue. I just upgraded my pisignage server to the latest release, but I'm still not able to upgrade the player to 5.0.4. Looking at the log file, the upgrade is looking for file piimage5.0.4-p2-v20.zip, but when I browse the files on the server, the closest file version is piimage5.0.4-p2-v14.zip. 

I suppose I could rename the file to what the upgrade is looking for, but I'm not sure that is a good idea.

[forever.log](https://github.com/colloqi/piSignage/files/13880699/forever.log)

Any help would be appreciated.

Thanks!

colloqi, 2024-01-10:
Thanks! latest release needs -v20 files for node v20 and it was not added to OS release. Just now released an updated version. Can you check the same?

jacobmessner, 2024-01-10:
Thanks for the quick response! 

I have updated the server again and it did indeed download the -v20 file. However, after updating the player to 5.0.4, the mouse cursor is still displayed in the middle of the screen.

Noddley, 2024-01-10:
repulled the Github and updated my server. updated Pi's, everything seems to be working! mouse courser are off the screen for me. anyways thanks for the help!

jacobmessner, 2024-01-11:
Is it possible the fix wasn't pulled into version 5.04? Or is it possible to downgrade to 5.0.3?

SCAI-Factory, 2024-01-11:
Confirming that the cursor was present with Raspberry Pi 5 running piSignage 5.01..  Upon a sucessful update to piSignage 5.0.4, the cursor is removed/hidden.

Noddley, 2024-01-11:
From what I have experienced, my 4 raspberry pi 4s the cursor was stuck in screen prior to the 5.0.4 update, once the new node files were in the server files. Then we could successfully update. After that for me the cursor went away but @jacobmessner is still claiming that he is having issues.

SCAI-Factory, 2024-01-11:
@Noddley yeah I think I worded my comment incorrectly.  I guess what I meant to say was that the cursor was removed/hidden on my Raspberry Pi 5 once we also updated the server and performed a successful update to piSignage 5.0.4.  I've restarted the Pi 5 multiple times just to test and the cursor remains hidden.

I will say, however, that when I connect my wireless mouse up via USB and move the mouse the cursor is visible again.  I suppose this is to be expected.

Noddley, 2024-01-11:
@SCAI-Factory, my bad. I see that now but if you want to remove the ability to use usb ports there a get guide at this link: https://help.pisignage.com/hc/en-us/articles/360012177871-Disable-mouse-and-keyboard-inputs-for-player-from-server

jacobmessner, 2024-01-11:
Not sure if it helps, but if I disconnect and reconnect the HDMI cable after the Pi is on, then the cursor jumps way off to the right side of the screen so its barely visible (like a single line of white pixels). But after a reboot, the cursor re-centers itself.

Also, I am using the 8gb ram model of raspberry pi 5.

colloqi, 2024-01-14:
https://forums.raspberrypi.com/viewtopic.php?t=358285#p2175588, this was used in 5.0.4 update

jacobmessner, 2024-01-17:
We have our players segmented from the outside internet. I wonder if that is causing problems with getting the cursor fix. I thought the server would download everything needed for each player and then serve as needed so that the players wouldn't need to download things individually.

colloqi, 2024-01-18:
It is like that, what does your player show version as and do you see any errors in /home/pi/forever.log?

Alternatively we have now 5.0.4 SD card image available which has these changes incorporated.

colloqi, 2024-01-19:
One option is to take the new 5.0.5 SD card image and reprogram the SD card

jacobmessner, 2024-01-30:
I have updated all of our players to 5.0.5 and the cursor now disappears appropriately. I would consider this issue closed. 

Thank you to everyone that helped!

closed 2024-01-30 by Noddley